stupidloud/nanopi-openwrt #924: 更新脚本无法更新固件，使用过很多个固件都没更新成功过，每次最后都是404

beautifulangle, 2021-11-01:
换过很多个固件，每次最后都是404，不知道是不是我不会操作问题

> 

root@MainRoute:~# wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | sh

Downloading https://mirrors.tencent.com/lede/snapshots/targets/x86/64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_core
Downloading https://mirrors.tencent.com/lede/snapshots/packages/x86_64/base/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_base
Downloading https://mirrors.tencent.com/lede/snapshots/packages/x86_64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_packages
Downloading https://mirrors.tencent.com/lede/snapshots/packages/x86_64/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_luci
Downloading https://mirrors.tencent.com/lede/snapshots/packages/x86_64/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_routing
Downloading https://mirrors.tencent.com/lede/snapshots/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_telephony
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Package coreutils-truncate (8.32-6) installed in root is up to date.
--2021-10-31 18:04:32--  https://ghproxy.com/https://raw.githubusercontent.com/klever1988/nanopi-openwrt/zstd-bin/truncate
Resolving ghproxy.com... 132.226.169.65
Connecting to ghproxy.com|132.226.169.65|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 45177 (44K) [application/octet-stream]
Saving to: '/tmp/truncate'

truncate                   100%[=======================================>]  44.12K  92.0KB/s    in 0.5s    

Last-modified header missing -- time-stamps turned off.
2021-10-31 18:04:36 (92.0 KB/s) - '/tmp/truncate' saved [45177/45177]

--2021-10-31 18:04:36--  https://ghproxy.com/https://raw.githubusercontent.com/klever1988/nanopi-openwrt/zstd-bin/ddnz
Resolving ghproxy.com... 146.56.175.58
Connecting to ghproxy.com|146.56.175.58|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 17936 (18K) [application/octet-stream]
Saving to: '/tmp/ddnz'

ddnz                       100%[=======================================>]  17.52K   101KB/s    in 0.2s    

Last-modified header missing -- time-stamps turned off.
2021-10-31 18:04:39 (101 KB/s) - '/tmp/ddnz' saved [17936/17936]

--2021-10-31 18:04:39--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-10-31/vmware-inc-vmware7-1.img.gz
Resolving ghproxy.com... 146.56.175.58
Connecting to ghproxy.com|146.56.175.58|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2021-10-31 18:04:42 ERROR 404: NOT FOUND.

gzip: invalid magic
今天的固件还没更新，尝试下载昨天的固件
--2021-10-31 18:04:42--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-10-30/vmware-inc-vmware7-1.img.gz
Resolving ghproxy.com... 146.56.175.58
Connecting to ghproxy.com|146.56.175.58|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2021-10-31 18:04:45 ERROR 404: NOT FOUND.

gzip: invalid magic
没找到最新的固件，脚本退出
root@MainRoute:~# 
root@MainRoute:~#

klever1988, 2021-11-01:
修了一下升级脚本

你再试一下看看

beautifulangle, 2021-11-08:
> 修了一下升级脚本
> 
> 你再试一下看看

还是不行，我试了4号，7号的都不能更新
他是这样的

root@ImmortalWrt:~# wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | sh
Package coreutils-truncate (8.32-6) installed in root is up to date.
--2021-11-08 14:59:52--  https://ghproxy.com/https://raw.githubusercontent.com/klever1988/nanopi-openwrt/zstd-bin/truncate
Resolving ghproxy.com... 132.226.169.65
Connecting to ghproxy.com|132.226.169.65|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 45177 (44K) [application/octet-stream]
Saving to: '/tmp/truncate'

truncate                               100%[============================================================================>]  44.12K   231KB/s    in 0.2s    

Last-modified header missing -- time-stamps turned off.
2021-11-08 14:59:56 (231 KB/s) - '/tmp/truncate' saved [45177/45177]

--2021-11-08 14:59:56--  https://ghproxy.com/https://raw.githubusercontent.com/klever1988/nanopi-openwrt/zstd-bin/ddnz
Resolving ghproxy.com... 132.226.169.65
Connecting to ghproxy.com|132.226.169.65|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 17936 (18K) [application/octet-stream]
Saving to: '/tmp/ddnz'

ddnz                                   100%[============================================================================>]  17.52K  --.-KB/s    in 0s      

Last-modified header missing -- time-stamps turned off.
2021-11-08 15:00:02 (276 MB/s) - '/tmp/ddnz' saved [17936/17936]

--2021-11-08 15:00:02--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-11-08/x86.img.gz
Resolving ghproxy.com... 132.226.169.65
Connecting to ghproxy.com|132.226.169.65|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 120681565 (115M) [application/octet-stream]
Saving to: 'STDOUT'

-                                      100%[============================================================================>] 115.09M  1.27MB/s    in 87s     

2021-11-08 15:01:45 (1.32 MB/s) - written to stdout [120681565/120681565]

--2021-11-08 15:01:45--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-11-08/x86.img.md5
Resolving ghproxy.com... 132.226.169.65
Connecting to ghproxy.com|132.226.169.65|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 42 [application/octet-stream]
Saving to: 'md5sum.txt'

md5sum.txt                             100%[============================================================================>]      42  --.-KB/s    in 0s      

2021-11-08 15:01:48 (20.3 MB/s) - 'md5sum.txt' saved [42/42]

今天固件已下载，准备解压
GPT PMBR size mismatch (1262111 != 1262142) will be corrected by write.
The backup GPT table is corrupt, but the primary appears OK, so that will be used.
The backup GPT table is not on the end of the device. This problem will be corrected by write.
Checking that no-one is using this disk right now ... OK

Disk FriendlyWrt.img: 616.28 MiB, 646217216 bytes, 1262143 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: ABF6A8A4-7813-BEC7-0596-1C6F4AE99D00

Old situation:

Device             Start     End Sectors  Size Type
FriendlyWrt.img1     512   33279   32768   16M Linux filesystem
FriendlyWrt.img2   33280 1262079 1228800  600M Linux filesystem
FriendlyWrt.img128    34     511     478  239K BIOS boot

Partition table entries are not in disk order.

FriendlyWrt.img2: 
New situation:
Disklabel type: gpt
Disk identifier: ABF6A8A4-7813-BEC7-0596-1C6F4AE99D00

Device             Start     End Sectors  Size Type
FriendlyWrt.img1     512   33279   32768   16M Linux filesystem
FriendlyWrt.img2   33280 1262109 1228830  600M Linux filesystem
FriendlyWrt.img128    34     511     478  239K BIOS boot

Partition table entries are not in disk order.

The partition table has been altered.
Syncing disks.
解压已完成，准备编辑镜像文件，写入备份信息
Mon Nov  8 15:02:00 CST 2021 upgrade: Saving config files...
注意：由于已知的问题，“网络接口”配置无法继承，重启后需要重新设置WAN拨号和LAN网段信息
备份文件已经写入，移除挂载
e2fsck 1.45.6 (20-Mar-2020)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
Padding at end of inode bitmap is not set. Fix? yes

rootfs: ***** FILE SYSTEM WAS MODIFIED *****
rootfs: 8584/38400 files (0.0% non-contiguous), 75810/153600 blocks
resize2fs 1.45.6 (20-Mar-2020)
Resizing the filesystem on /dev/loop0p2 to 153603 (4k) blocks.
The filesystem on /dev/loop0p2 is now 153603 (4k) blocks long.

正在打包...
开始写入，请勿中断...
../ddnz: line 1: ELF
                    @@P?@@@@@@@@@@++..A.A: not found
../ddnz: line 1: syntax error: unterminated quoted string

klever1988, 2021-11-09:
> 还是不行，我试了4号，7号的都不能更新 他是这样的
>
> root@ImmortalWrt:~# wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | sh Package coreutils-truncate (8.32-6) installed in root is [...]

试一下
```wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | bash```

beautifulangle, 2021-11-10:
> 试一下 `wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | bash`

感谢大大，用这个命令升级成功了