stupidloud/nanopi-openwrt #930: 今天更新的固件有问题

mengxinz, 2021-11-04:
@klever1988 更新完后无线网卡没反应了，插网线也无法进去后台，slim版

klever1988, 2021-11-04:
重刷呢？

mengxinz, 2021-11-04:
> 重刷呢？

重刷能进后台，插上无线网卡就完全不能用了

klever1988, 2021-11-04:
插上之后内核日志发过来看看啊

klever1988, 2021-11-04:
插上网卡再拔掉 有线连接还能连上不
可以话复制内核日志过来瞅瞅

mengxinz, 2021-11-04:
> 插上网卡再拔掉 有线连接还能连上不 可以话复制内核日志过来瞅瞅

必须拔掉电源重启

klever1988, 2021-11-04:
什么网卡 知道调用的时哪个驱动不？ 你可以用以前能用的固件去看内核日志

mengxinz, 2021-11-04:
> 什么网卡 知道调用的时哪个驱动不？ 你可以用以前能用的固件去看内核日志

[image]

mengxinz, 2021-11-04:
> > 什么网卡 知道调用的时哪个驱动不？ 你可以用以前能用的固件去看内核日志
> 
> [image]

[image]

mengxinz, 2021-11-04:
[image]

klever1988, 2021-11-04:
这里面也没有加在无线网卡驱动的片段

mengxinz, 2021-11-04:
> 这里面也没有加载无线网卡驱动的片段

应该是rtl8811cu,之前都能正常使用升级完就不能用了重刷插上网卡也炸，只能回退之前的了

klever1988, 2021-11-04:
你可以不带网卡启动
启动完成后再插入网卡
再用dmesg打印后半段加载驱动的内核日志

mengxinz, 2021-11-04:
> 你可以不带网卡启动 启动完成后再插入网卡 再用dmesg打印后半段加载驱动的内核日志

我回退了，现在正常使用等别人来反馈吧

klever1988, 2021-11-04:
可以的~

rainplus, 2021-11-07:
同样遇到这个问题. .. 
网卡:
comfast 811ac

只要插上网卡就死机了.. 动不了

mengxinz, 2021-11-07:
> 版本: r2s-slim.img.gz 11/06
> 
> 同样遇到这个问题. .. 网卡: comfast 811ac
> 
> 只要插上网卡就死机了.. 动不了
> 
> ```
> [  432.819762] usb 1-1: new high-speed USB device number 3 using ehci-platform
> [  432.978058] RTW: [HALMAC]11692M
> [  432.978058] HALMAC_MAJOR_VER = 1
> [  432.978058] HALMAC_PROTOTYPE_VER = 4
> [  432.978058] HALMAC_MINOR_VER = 19
> [  432.978058] HALMAC_PATCH_VER = 3
> [  432.981108] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
> [  433.253463] RTW: HW EFUSE
> [  433.253737] RTW: 0x000: 29 81 00 BC  09 00 21 00  6E 04 A4 10  10 00 30 0B
> [  433.254492] RTW: 0x010: 22 22 22 22  22 22 2A 2A  2A 2A 2A 02  FF FF FF FF
> [  433.255244] RTW: 0x020: FF FF 27 27  26 26 26 26  26 26 26 26  25 25 25 25
> [  433.256058] RTW: 0x030: 24 FF FF FF  FF FF D0 FF  FF FF 22 22  22 22 22 22
> [  433.257125] RTW: 0x040: 2A 2A 2A 2A  2A 02 FF FF  FF FF FF FF  FF FF FF FF
> [  433.257877] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.258628] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.259380] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.260186] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.261237] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.261989] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.262741] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 2B 1A 00  FF FF FF FF
> [  433.263512] RTW: 0x0C0: FF 22 00 41  00 00 00 00  00 FF 22 FF  FF FF FF FF
> [  433.264605] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.265357] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.266108] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.266861] RTW: 0x100: DA 0B 11 C8  E3 46 02 20  0D B0 41 5B  DC 09 03 52
> [  433.267656] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63
> [  433.268713] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF
> [  433.269465] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.270216] RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF
> [  433.270968] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.271764] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.272824] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.273576] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.274327] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.275090] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.275884] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.276945] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.277697] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.278449] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.279200] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
> [  433.280379] RTW: hal_com_config_channel_plan chplan:0x7F
> [  433.294742] RTW: [HALMAC][ALWAYS]shall R reg twice!!
> [  433.298881] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
> [  433.413374] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
> [  433.414141] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
> [  433.414885] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
> [  433.415825] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
> [  433.416639] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
> [  433.417385] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
> [  433.418133] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
> [  433.420157] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
> [  433.424767] RTW: rtw_ndev_init(wlan0) if1 mac_addr=20:0d:b0:41:5b:dc
> [  433.426972] RTW: rtw_ndev_init(wlan1) if2 mac_addr=22:0d:b0:41:5b:dc
> [  433.429855] reg-fixed-voltage vcc-rtl8153-regulator: deferred probe timeout, ignoring dependency
> ```

@klever1988 有反馈了

klever1988, 2021-11-07:
我解决不了 这看起来像是驱动和内核不兼容
等上游提供什么patch之类的吧

mengxinz, 2021-11-07:
> 我解决不了 这看起来像是驱动和内核不兼容 等上游提供什么patch之类的吧
行吧什么时候修复了说明一下，还有就是那个风扇脚本好像也有问题一直转不会停你看下

klever1988, 2021-11-07:
风扇控制换成了更新的版本
你可以先看看系统日志看看脚本是否正常运行（系统日志处搜索pwfan)
温度梯度控制如下
https://github.com/friendlyarm/friendlywrt/blob/32cf7e149db3fd811cffb50930ae02c2defd80a0/target/linux/rockchip/armv8/base-files/usr/bin/fa-fancontrol-direct.sh#L36

你可以直接修改路由器的/usr/bin/fa-fancontrol-direct.sh这个文件

mengxinz, 2021-11-10:
> 我解决不了 这看起来像是驱动和内核不兼容 等上游提供什么patch之类的吧

大佬网卡问题现在也没有解决，你向相关的上游反馈一下吧

closed 2021-11-12 by mengxinz

AugustHero, 2021-12-04:
老哥，你现在用什么版本呢？

yi-ge, 2022-05-18:
现在仍然有这个问题