stupidloud/nanopi-openwrt #934: 研究COMFAST 811AC 无法使用的可能性

FirYuen, 2021-11-10:
我最近买了用RTL8821CU芯片，COMFAST 811AC ,如果插着开机,则wifi无法使用,具体现象是 ifconfig卡死,iwinfo卡死,
我以为是immortalwrt的问题,所以提了[issue](https://github.com/immortalwrt/immortalwrt/issues/504#issue-1045431294),
得到的回复是猜测可能是kernel 5.10的问题,我研究本项目的编译方式时发现本项目可能是使用https://github.com/klever1988/lede 编译了基础的package和内核等内容,随后我直接使用https://github.com/klever1988/lede编译了r2s 5.10内核版本的固件,发现COMFAST 811AC WiFi使用正常,我再看发现是不是在编译r2s固件时,使用了lede项目的package缓存,但是用了immortalwrt的源码导致两边出现了兼容性问题?
https://github.com/klever1988/nanopi-openwrt/blob/95437e746f66af86ecc938b8144852102a18c34d/.github/workflows/lo-test.yml#L47-L60

mengxinz, 2021-11-10:
> 我最近买了用RTL8821CU芯片，COMFAST 811AC ,如果插着开机,则wifi无法使用,具体现象是 ifconfig卡死,iwinfo卡死, 我以为是immortalwrt的问题,所以提了[issue](https://github.com/immortalwrt/immortalwrt/issues/504#issue-1045431294), 得到的回复是猜测可能是kernel [...]

@klever1988 大佬快来又有新的反馈了

klever1988, 2021-11-10:
L大的rockchip内核是5.4，所以没有这个bug
    我把内核退到5.4吧

FirYuen, 2021-11-10:
> L大的rockchip内核是5.4，所以没有这个bug 我把内核退到5.4吧
我有两个疑问想问一下哈:
1. 如果像我理解的那样,编译时使用了L大5.4内核的package缓存,实际编译时又用了immortal 的5.10内核的代码,这种内核不匹配的情况我理解应该无法编译成功啊? 毕竟想要安装的kmod和当前内核版本不匹配啊
2. 如果说不回退内核直接修改https://github.com/klever1988/lede的github action编译出5.10的缓存包,是不是这个问题就解决了呢? 因为我经过测试,直接编译https://github.com/klever1988/lede 5.10版本的内核是可以正常工作的

mengxinz, 2021-11-10:
> 1. 如果像我理解的那样,编译时使用了L大5.4内核的package缓存,实际编译时又用了immortal 的5.10内核的代码,这种内核不匹配的情况我理解应该无法编译成功啊? 毕竟想要安装的kmod和当前内核版本不匹配啊
> 2. [...]

这个办法好，大佬试一下 @klever1988

mengxinz, 2021-11-11:
> L大的rockchip内核是5.4，所以没有这个bug 我把内核退到5.4吧
@klever1988 更新了今天的固件问题依旧没解决

FirYuen, 2021-11-12:
最新的更新 使用5.4的内核问题仍然存在

mengxinz, 2021-11-12:
> 最新的更新 使用5.4的内核问题仍然存在

@klever1988 大家都反馈问题并没有修复，说不定真是编译出了问题，建议借鉴一下楼上所说的编译方式

FirYuen, 2021-11-12:
> > 最新的更新 使用5.4的内核问题仍然存在
> 
> @klever1988 大家都反馈问题并没有修复，说不定真是编译出了问题，建议借鉴一下楼上所说的编译方式

今天使用L大的kernel 5.4版本源码编译一开始使用正常,但是某些操作后会出现同样的现象,所以问题应该来源于上游,与@klever1988 应该无关,这个问题关闭了,后续我会继续进行尝试.

closed 2021-11-12 by FirYuen

novicewu, 2022-04-24:
大佬们这个问题解决了吗？我今天买了CF-811AC 插R2S也发现了一模一样的问题。折腾一天换了各种固件都没搞定。有谁可以发一个可用的固件？