stupidloud/nanopi-openwrt #943: DDNS报错GNU Wget Error: '4'

lfdfxl, 2021-11-19:
DDNS使用he.net，一直报错GNU Wget Error: '4'。重启服务、勾选使用curl也无法解决，请大佬帮忙看看

Fri Nov 19 09:43:17 2021 user.notice ddns-scripts[7994]: freefly666: PID '7994' started at 2021-11-19 09:43
Fri Nov 19 09:43:22 2021 user.err ddns-scripts[7994]: freefly666: GNU Wget Error: '4'
Fri Nov 19 09:43:22 2021 user.warn ddns-scripts[7994]: freefly666: Transfer failed - retry 1/0 in 60 seconds
Fri Nov 19 09:44:26 2021 user.err ddns-scripts[7994]: freefly666: GNU Wget Error: '4'
Fri Nov 19 09:44:26 2021 user.warn ddns-scripts[7994]: freefly666: Transfer failed - retry 2/0 in 60 seconds
Fri Nov 19 09:45:31 2021 user.err ddns-scripts[7994]: freefly666: GNU Wget Error: '4'
Fri Nov 19 09:45:31 2021 user.warn ddns-scripts[7994]: freefly666: Transfer failed - retry 3/0 in 60 seconds
Fri Nov 19 09:45:59 2021 user.notice ddns-scripts[7994]: freefly666: PID '7994' terminated by 'SIGTERM' at 2021-11-19 09:45
Fri Nov 19 09:46:00 2021 user.notice ddns-scripts[9432]: freefly666: PID '9432' started at 2021-11-19 09:46
Fri Nov 19 09:46:05 2021 user.err ddns-scripts[9432]: freefly666: GNU Wget Error: '4'
Fri Nov 19 09:46:05 2021 user.warn ddns-scripts[9432]: freefly666: Transfer failed - retry 1/0 in 60 seconds
Fri Nov 19 09:47:09 2021 user.err ddns-scripts[9432]: freefly666: GNU Wget Error: '4'
Fri Nov 19 09:47:09 2021 user.warn ddns-scripts[9432]: freefly666: Transfer failed - retry 2/0 in 60 seconds
Fri Nov 19 09:48:13 2021 user.err ddns-scripts[9432]: freefly666: GNU Wget Error: '4'
Fri Nov 19 09:48:13 2021 user.warn ddns-scripts[9432]: freefly666: Transfer failed - retry 3/0 in 60 seconds
Fri Nov 19 09:49:17 2021 user.err ddns-scripts[9432]: freefly666: GNU Wget Error: '4'
Fri Nov 19 09:49:17 2021 user.warn ddns-scripts[9432]: freefly666: Transfer failed - retry 4/0 in 60 seconds
Fri Nov 19 09:50:21 2021 user.err ddns-scripts[9432]: freefly666: GNU Wget Error: '4'
Fri Nov 19 09:50:21 2021 user.warn ddns-scripts[9432]: freefly666: Transfer failed - retry 5/0 in 60 seconds
Fri Nov 19 09:51:26 2021 user.err ddns-scripts[9432]: freefly666: GNU Wget Error: '4'
Fri Nov 19 09:51:26 2021 user.warn ddns-scripts[9432]: freefly666: Transfer failed - retry 6/0 in 60 seconds
Fri Nov 19 09:51:33 2021 user.notice ddns-scripts[11591]: wifi_freefly666: PID '11591' started at 2021-11-19 09:51
Fri Nov 19 09:52:30 2021 user.err ddns-scripts[9432]: freefly666: GNU Wget Error: '4'
Fri Nov 19 09:52:30 2021 user.warn ddns-scripts[9432]: freefly666: Transfer failed - retry 7/0 in 60 seconds
Fri Nov 19 09:53:34 2021 user.err ddns-scripts[9432]: freefly666: GNU Wget Error: '4'
Fri Nov 19 09:53:34 2021 user.warn ddns-scripts[9432]: freefly666: Transfer failed - retry 8/0 in 60 seconds
Fri Nov 19 09:53:43 2021 user.err ddns-scripts[11591]: wifi_freefly666: cURL Error: '28'
Fri Nov 19 09:53:43 2021 user.warn ddns-scripts[11591]: wifi_freefly666: Transfer failed - retry 1/0 in 60 seconds
Fri Nov 19 09:54:39 2021 user.err ddns-scripts[9432]: freefly666: GNU Wget Error: '4'
Fri Nov 19 09:54:39 2021 user.warn ddns-scripts[9432]: freefly666: Transfer failed - retry 9/0 in 60 seconds

klever1988, 2021-11-19:
https://www.gnu.org/software/wget/manual/html_node/Exit-Status.html

lfdfxl, 2021-11-20:
> https://www.gnu.org/software/wget/manual/html_node/Exit-Status.html

感谢，已找到原因。
ddns子域名和主域名需要用@分割，不能用.

closed 2021-11-20 by lfdfxl