stupidloud/nanopi-openwrt #948: R2S 这几天的固件全都是 风扇转个不停

SZFtutian, 2021-11-21:
风扇脚本都改了也没有 才疏学浅 请大神给个直接可以用的

iscodeeasyornot, 2021-11-21:
`https://github.com/friendlyarm/friendlywrt/raw/master-v19.07.1/target/linux/rockchip-rk3328/base-files/usr/bin/start-rk3328-pwm-fan.sh`
把这个脚本放 /etc/rc.local里

SZFtutian, 2021-11-21:
> `https://github.com/friendlyarm/friendlywrt/raw/master-v19.07.1/target/linux/rockchip-rk3328/base-files/usr/bin/start-rk3328-pwm-fan.sh` 把这个脚本放 /etc/rc.local 里

操作了没有用

iscodeeasyornot, 2021-11-21:
> > `https://github.com/friendlyarm/friendlywrt/raw/master-v19.07.1/target/linux/rockchip-rk3328/base-files/usr/bin/start-rk3328-pwm-fan.sh` 把这个脚本放 /etc/rc.local 里
> 
> 操作了没有用

加bash了么？

SZFtutian, 2021-11-22:
> > > `https://github.com/friendlyarm/friendlywrt/raw/master-v19.07.1/target/linux/rockchip-rk3328/base-files/usr/bin/start-rk3328-pwm-fan.sh` 把这个脚本放 /etc/rc.local 里
> > 
> > 
> > 操作了没有用
> 
> 加bash了么？
没有用刷回以前的固件了

SZFtutian, 2021-11-22:
> > > `https://github.com/friendlyarm/friendlywrt/raw/master-v19.07.1/target/linux/rockchip-rk3328/base-files/usr/bin/start-rk3328-pwm-fan.sh` 把这个脚本放 /etc/rc.local 里
> > 
> > 
> > 操作了没有用
> 
> 加bash了么？

没有用刷回以前的版本了

ks0998-5431, 2021-11-22:
> > > > `https://github.com/friendlyarm/friendlywrt/raw/master-v19.07.1/target/linux/rockchip-rk3328/base-files/usr/bin/start-rk3328-pwm-fan.sh` 把这个脚本放 /etc/rc.local 里
> > > 
> > > 
> > > 操作了没有用
> > 
> > 
> > 加bash了么？
> 
> 没有用刷回以前的版本了

哪天的版本，我最近也是刷的我焦头烂额

SZFtutian, 2021-11-22:
> > > > > `https://github.com/friendlyarm/friendlywrt/raw/master-v19.07.1/target/linux/rockchip-rk3328/base-files/usr/bin/start-rk3328-pwm-fan.sh` 把这个脚本放 /etc/rc.local 里
> > > > 
> > > > 
> > > > 操作了没有用
> > > 
> > > 
> > > 加bash了么？
> > 
> > 
> > 没有用刷回以前的版本了
> 
> 哪天的版本，我最近也是刷的我焦头烂额

刷的11月2号的

klever1988, 2021-11-22:
今天的版本做了改动，应该可以修复了