stupidloud/nanopi-openwrt #952: R1 Plus的WAN口不能识别网卡

litao1791, 2021-11-22:
lan口正常，wan口不正常，应该是驱动有问题

klever1988, 2021-11-22:
r1p和r2s硬件资源是一样的 你可以试一下他们的官方固件看看网卡有没有问题

John-SadJoe-Doe, 2021-11-23:
> r1p和r2s硬件资源是一样的 你可以试一下他们的官方固件看看网卡有没有问题

我也用的是r1p   wan口无法识别是最近的固件出现的    我现在回退到11.2的固件  wan口正常识别     但如果切换主题可能会出现wan口无法拨号或者如图所示的问题   并且再切换回默认主题无法解决  需要重刷固件
[image]

IPlayForHer, 2021-11-27:
Thu Jan 21 17:00:10 2016 daemon.notice netifd: Interface 'wan' is disabled
Thu Jan 21 17:00:11 2016 kern.info kernel: [  112.137612] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:03] driver [YT8521 Ethernet]
Thu Jan 21 17:00:11 2016 daemon.notice netifd: Interface 'wan' is enabled
Thu Jan 21 17:00:11 2016 kern.info kernel: [  112.151890] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
Thu Jan 21 17:00:11 2016 kern.warn kernel: [  112.152628] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
Thu Jan 21 17:00:11 2016 kern.info kernel: [  112.153244] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode

158212892, 2021-11-27:
我也是wan口一直不通，有解决方案了么😭😭

IPlayForHer, 2021-11-27:
> > r1p和r2s硬件资源是一样的 你可以试一下他们的官方固件看看网卡有没有问题
> 
> 
> 
> 我也用的是r1p   wan口无法识别是最近的固件出现的    我现在回退到11.2的固件  wan口正常识别     但如果切换主题可能会出现wan口无法拨号或者如图所示的问题   并且再切换回默认主题无法解决  需要重刷固件
> 
> [image]
> 
> 
> 
> 

你试试用手机打开是否有图片的问题

IPlayForHer, 2021-11-27:
> 我也是wan口一直不通，有解决方案了么😭😭

11月2号的固件是可以使用的

158212892, 2021-11-27:
> > 我也是wan口一直不通，有解决方案了么sobsob
> 
> 11月2号的固件是可以使用的

我在用10.1的，能用就行，就是不能更新新版了

AmadeusGhost, 2021-12-07:
> Thu Jan 21 17:00:11 2016 kern.info kernel: [  112.137612] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:03] driver [YT8521 Ethernet]
> 相同问题，这个是网卡日志，之前用的2021.10.1的固件没有问题更新之后出错的，重刷了几遍还是连不通。

你这个是 r2c 的吧...

radiation-chan, 2022-01-18:
想问一下，现在解决了吗？我刷了2022.1.12的还是不行。

litao1791, 2022-01-18:
我自己编译的，如果需要，我可以把我的给你

radiation-chan, 2022-01-18:
> 我自己编译的，如果需要，我可以把我的给你

可以的话发一下链接，谢谢，我还没找到其他合适的固件。T T

litao1791, 2022-01-18:
链接: https://pan.baidu.com/s/1n7DPjbGWsFg7WZZCxdYesA?pwd=54ih 提取码: 54ih 复制这段内容后打开百度网盘手机App，操作更方便哦