stupidloud/nanopi-openwrt #953: 自动更新脚本在docker版无法使用

xiaowuap, 2021-11-25:
t@ImmortalWrt:~# wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | ver=-docker sh
Downloading https://mirrors.tencent.com/lede/snapshots/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_core
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_base
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_packages
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_luci
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_routing
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_telephony
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Package coreutils-truncate (8.32-6) installed in root is up to date.
--2021-11-25 19:31:07--  https://ghproxy.com/https://raw.githubusercontent.com/klever1988/nanopi-openwrt/zstd-bin/truncate
Resolving ghproxy.com... 132.226.169.65
Connecting to ghproxy.com|132.226.169.65|:443... connected.
HTTP request sent, awaiting response... 302 FOUND
Location: https://cdn.jsdelivr.net/gh/klever1988/nanopi-openwrt@zstd-bin/truncate [following]
--2021-11-25 19:31:08--  https://cdn.jsdelivr.net/gh/klever1988/nanopi-openwrt@zstd-bin/truncate
Resolving cdn.jsdelivr.net... 240e:97e:2000:c108::9, 59.56.26.18
Connecting to cdn.jsdelivr.net|240e:97e:2000:c108::9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 45177 (44K) [application/octet-stream]
Saving to: '/tmp/truncate'

truncate                                          100%[==========================================================================================================>]  44.12K  --.-KB/s    in 0.001s  

Last-modified header missing -- time-stamps turned off.
2021-11-25 19:31:08 (30.4 MB/s) - '/tmp/truncate' saved [45177/45177]

--2021-11-25 19:31:08--  https://ghproxy.com/https://raw.githubusercontent.com/klever1988/nanopi-openwrt/zstd-bin/ddnz
Resolving ghproxy.com... 132.226.169.65
Connecting to ghproxy.com|132.226.169.65|:443... connected.
HTTP request sent, awaiting response... 302 FOUND
Location: https://cdn.jsdelivr.net/gh/klever1988/nanopi-openwrt@zstd-bin/ddnz [following]
--2021-11-25 19:31:08--  https://cdn.jsdelivr.net/gh/klever1988/nanopi-openwrt@zstd-bin/ddnz
Resolving cdn.jsdelivr.net... 240e:97e:2000:c108::9, 59.56.26.18
Connecting to cdn.jsdelivr.net|240e:97e:2000:c108::9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 17936 (18K) [application/octet-stream]
Saving to: '/tmp/ddnz'

ddnz                                              100%[==========================================================================================================>]  17.52K  --.-KB/s    in 0s      

Last-modified header missing -- time-stamps turned off.
2021-11-25 19:31:08 (36.9 MB/s) - '/tmp/ddnz' saved [17936/17936]

--2021-11-25 19:31:08--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-11-25/r4s-docker.img.gz
Resolving ghproxy.com... 132.226.169.65
Connecting to ghproxy.com|132.226.169.65|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2021-11-25 19:31:09 ERROR 404: NOT FOUND.

gzip: invalid magic
今天的固件还没更新，尝试下载昨天的固件
--2021-11-25 19:31:09--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-11-24/r4s-docker.img.gz
Resolving ghproxy.com... 132.226.169.65
Connecting to ghproxy.com|132.226.169.65|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2021-11-25 19:31:09 ERROR 404: NOT FOUND.

gzip: invalid magic
没找到最新的固件，脚本退出
root@ImmortalWrt:~# 
这里找的固件名似乎不对？

klever1988, 2021-11-25:
[image]

因为今天还没更新，等一会吧

xiaowuap, 2021-11-27:
问题不应该是新固件的文件名变成了r4s-with-docker.img.gz，而不是r4s-docker.img.gz了吗

klever1988, 2021-11-27:
wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | ver=-with-docker sh