stupidloud/nanopi-openwrt #956: 无法调度: /cgi-bin/luci/admin/system/fstab

lcl-101, 2021-11-28:
[image]
smb挂载有问题，点击挂载路径跳转找不到这个路径

klever1988, 2021-11-28:
不知道你指的是哪个插件，cifs-mount我没有问题

[image]

lcl-101, 2021-11-29:
> 不知道你指的是哪个插件，cifs-mount我没有问题
> 
> [image]

[image]
是这样d的我改smb挂载里点击挂载路径，会跳到/cgi-bin/luci/admin/system/fstab这个地址，
但挂载点实际上是 cgi-bin/luci/admin/system/mounts 这地址

[image]

我不清楚是不是还缺什么包需要安装

klever1988, 2021-11-29:
这个应该是插件的bug，改一下就行

lcl-101, 2021-11-29:
> 这个应该是插件的bug，改一下就行

嗯嗯，我这边下的r4s，slim这个版本，安装包都是本地的，到时候，我怎么单独更新这个包呀，编译这块我还是个小白，刚入坑不久

klever1988, 2021-11-29:
/usr/lib/lua/luci/model/cbi/cifs.lua
你只要修改上面这个文件， fstab改成mounts就可以了

lcl-101, 2021-11-29:
> mounts

哈哈，搞定了，我们有没有群什么的？