stupidloud/nanopi-openwrt #957: 在爱快虚拟机装的，用了原来大大给的一个命令和这个命令都没法更新

beautifulangle, 2021-11-29:
MainRoute login: root
Password: 
Login incorrect
MainRoute login: root
Password: 
┌─────────────────────────────────────────────┐
│                                             │
│ mmmmm                         m       ""#   │
│   #   mmmmm mmmmm  mmm  mmm mm#mm  mmm  #   │
│   #   # # # # # # #" "# #" "  #   "   # #   │
│   #   # # # # # # #   # #     #   m"""# #   │
│ mm#mm # # # # # # "#m#" #     "mm "mm"# "mm │
│                                             │
│─────────────────────────────────────────────│
│            ImmortalWrt SNAPSHOT             │
└─────────────────────────────────────────────┘
root@MainRoute:~# wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | sh
Downloading https://mirrors.tencent.com/lede/snapshots/targets/x86/64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_core
Downloading https://mirrors.tencent.com/lede/snapshots/packages/x86_64/base/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_base
Downloading https://mirrors.tencent.com/lede/snapshots/packages/x86_64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_packages
Downloading https://mirrors.tencent.com/lede/snapshots/packages/x86_64/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_luci
Downloading https://mirrors.tencent.com/lede/snapshots/packages/x86_64/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_routing
Downloading https://mirrors.tencent.com/lede/snapshots/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_telephony
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Package coreutils-truncate (8.32-6) installed in root is up to date.
--2021-11-29 09:07:26--  https://ghproxy.com/https://raw.githubusercontent.com/klever1988/nanopi-openwrt/zstd-bin/truncate
Resolving ghproxy.com... 158.101.155.217
Connecting to ghproxy.com|158.101.155.217|:443... connected.
HTTP request sent, awaiting response... 302 FOUND
Location: https://cdn.jsdelivr.net/gh/klever1988/nanopi-openwrt@zstd-bin/truncate [following]
--2021-11-29 09:07:31--  https://cdn.jsdelivr.net/gh/klever1988/nanopi-openwrt@zstd-bin/truncate
Resolving cdn.jsdelivr.net... 118.123.233.23
Connecting to cdn.jsdelivr.net|118.123.233.23|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 45177 (44K) [application/octet-stream]
Saving to: '/tmp/truncate'

truncate                                                  100%[===================================================================================================================================>]  44.12K  --.-KB/s    in 0s      

Last-modified header missing -- time-stamps turned off.
2021-11-29 09:07:31 (136 MB/s) - '/tmp/truncate' saved [45177/45177]

--2021-11-29 09:07:31--  https://ghproxy.com/https://raw.githubusercontent.com/klever1988/nanopi-openwrt/zstd-bin/ddnz
Resolving ghproxy.com... 158.101.155.217
Connecting to ghproxy.com|158.101.155.217|:443... connected.
HTTP request sent, awaiting response... 302 FOUND
Location: https://cdn.jsdelivr.net/gh/klever1988/nanopi-openwrt@zstd-bin/ddnz [following]
--2021-11-29 09:07:34--  https://cdn.jsdelivr.net/gh/klever1988/nanopi-openwrt@zstd-bin/ddnz
Resolving cdn.jsdelivr.net... 118.123.233.23
Connecting to cdn.jsdelivr.net|118.123.233.23|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 17936 (18K) [application/octet-stream]
Saving to: '/tmp/ddnz'

ddnz                                                      100%[===================================================================================================================================>]  17.52K  --.-KB/s    in 0.001s  

Last-modified header missing -- time-stamps turned off.
2021-11-29 09:07:34 (33.1 MB/s) - '/tmp/ddnz' saved [17936/17936]

--2021-11-29 09:07:34--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-11-29/x86.img.gz
Resolving ghproxy.com... 158.101.155.217
Connecting to ghproxy.com|158.101.155.217|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2021-11-29 09:07:36 ERROR 404: NOT FOUND.

gzip: invalid magic
今天的固件还没更新，尝试下载昨天的固件
--2021-11-29 09:07:36--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-11-28/x86.img.gz
Resolving ghproxy.com... 158.101.155.217
Connecting to ghproxy.com|158.101.155.217|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 133561578 (127M) [application/octet-stream]
Saving to: 'STDOUT'

-                                                         100%[===================================================================================================================================>] 127.37M  2.00MB/s    in 71s     

2021-11-29 09:08:52 (1.79 MB/s) - written to stdout [133561578/133561578]

--2021-11-29 09:08:52--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-11-28/x86.img.md5
Resolving ghproxy.com... 158.101.155.217
Connecting to ghproxy.com|158.101.155.217|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 42 [application/octet-stream]
Saving to: 'md5sum.txt'

md5sum.txt                                                100%[===================================================================================================================================>]      42  --.-KB/s    in 0s      

2021-11-29 09:08:55 (23.4 MB/s) - 'md5sum.txt' saved [42/42]

昨天的固件已下载，准备解压
cat: can't open '/sys/block/sda/size': No such file or directory
expr: syntax error
root@MainRoute:~#

zhangguanzhang, 2021-12-22:
lsblk 看下输出

zhangguanzhang, 2021-12-22:
应该是vda