stupidloud/nanopi-openwrt #969: 这个系统可以在线升级吗？

adamzhou111, 2021-12-10:
[image]
这个老版本系统可以在线升级吗？经常性出现自动断网现象

adamzhou111, 2021-12-10:
大佬在吗？帮忙回复一下

klever1988, 2021-12-10:
我也不敢确定

再整个卡，克隆一下，升级失败插回去就是了

adamzhou111, 2021-12-10:
在新卡上安装新系统，能把旧卡的配置导入进去吗？配置导入应该是通用的吧？

adamzhou111, 2021-12-10:
> 我也不敢确定
> 
> 再整个卡，克隆一下，升级失败插回去就是了

在新卡上安装新系统，能把旧卡的配置导入进去吗？配置导入应该是通用的吧？

Zoooooooooo, 2021-12-10:
> 

大佬 我以为你的在线升级脚本是确认升级没问题呢，哈哈 昨晚弄了一下，目前没基础问题，就是插件丢了一两个。

SexyRose521, 2021-12-10:
https://github.com/klever1988/nanopi-openwrt仔细看看

klever1988, 2021-12-10:
> > 我也不敢确定
> > 再整个卡，克隆一下，升级失败插回去就是了
> 
> 在新卡上安装新系统，能把旧卡的配置导入进去吗？配置导入应该是通用的吧？

先克隆一下卡

klever1988, 2021-12-10:
> > 
> 
> 大佬 我以为你的在线升级脚本是确认升级没问题呢，哈哈 昨晚弄了一下，目前没基础问题，就是插件丢了一两个。

`wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | ver=-slim bash`

可以保留插件

Zoooooooooo, 2021-12-13:
> 

学习了，爱你。这参数在前面的说明里给出可以帮到更多人~

Zoooooooooo, 2021-12-13:
> > > 
> > 
> > 
> > 大佬 我以为你的在线升级脚本是确认升级没问题呢，哈哈 昨晚弄了一下，目前没基础问题，就是插件丢了一两个。
> 
> `wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | ver=-slim bash`
> 
> 可以保留插件

学习了，爱你。这参数在前面的说明里给出可以帮到更多人~