stupidloud/nanopi-openwrt #975: 希望更新下内核版本

gili-gili, 2021-12-13:
目前版本是5.4.162，有点老了导致一些东西装不上，希望能更新到5.10左右

klever1988, 2021-12-13:
啥玩意装不了

gili-gili, 2021-12-14:
> 啥玩意装不了

SQM，就是QoS，依赖装不上，至少要5.10，有办法自行升级吗

klever1988, 2021-12-14:
用slim是可以装的吧

gili-gili, 2021-12-14:
> 用slim是可以装的吧

不行，试过了，依赖要求要5.10，你看报错信息

已安装的软件包 kernel 版本不兼容，需要 5.10.83-1-dd3e905a… 而 5.4.162-1-545c66e6… 已安装。

要咋装啊......

klever1988, 2021-12-14:
可以的呀
[image]
[image]

gili-gili, 2021-12-15:
[image]
我下的是最新的镜像啊......奇了怪了

gili-gili, 2021-12-15:
> 可以的呀

你那个镜像内核版本是多少啊......

klever1988, 2021-12-15:
[image]

klever1988, 2021-12-15:
> > 可以的呀
> 
> 你那个镜像内核版本是多少啊......

你是启用了软件源里的http源吧，看看你下面这个地方的配置内容
[image]

gili-gili, 2021-12-15:
> 你是启用了软件源里的http源吧，看看你下面这个地方的配置内容 ![image]
对的我启用了......
所以说不能使用在线的，只能用本地的？

klever1988, 2021-12-15:
使用在线软件源的内核模块是无法兼容的
即使是你自己编译的固件
因为在线内核模块的编译生成摘要，每次编译都会不一样
即使内核版本一样，也会因为摘要不同，而提示不兼容

gili-gili, 2021-12-16:
> 使用在线软件源的内核模块是无法兼容的
> 即使是你自己编译的固件
> 因为内核模块的编译生成摘要，每次编译都会不一样
> 即使内核版本一样，也会因为摘要不同，而提示不兼容

懂了谢谢

closed 2021-12-16 by gili-gili