stupidloud/nanopi-openwrt #976: 【心态甭了】刷了12月所有的R2S的slim全都不分配ip地址

bingmiao, 2021-12-13:
之前用的是R2S-Minimal-2020-12-14-ROM
用了一年了，最近想升级试试，结果悲剧了，刷完不分配IP，必须手动才能进入管理界面。
其他设备连接后也是没有IP地址，固定内网IP后一切正常。

检查了一下所有DHCP相关设置都默认且正常，就是不分配IP

各位谁有办法帮我一下，感谢了

kgamecarter, 2021-12-14:
隔太久的版本更新會壞掉，整張卡重刷試試

najunuoyan, 2021-12-14:
我跟你也一样，刷了很多固件都是这现象，除了刚刷完固件开机第一次能正常获取到IP，之后就再也不行了

najunuoyan, 2021-12-14:
貌似跟你还不一样，我除了第一次刷机正常，再软重启或者断电重启，大概率LAN灯不会亮，即使电脑设置固定IP也没办法访问管理页面，WAN灯也有机率不会亮，心累啊。

zzusec, 2021-12-17:
> 之前用的是R2S-Minimal-2020-12-14-ROM
我之前也是用这个版本，一直很稳定，后来没有minimal版本，刷各种版本都有问题

najunuoyan, 2021-12-17:
我已经刷回R2S-Minimal-2020-12-14-ROM，这玩意儿稳定优先，功能多少是次要的

zzusec, 2021-12-17:
> 是次要的

这个版本我遇到一个问题，微信打开图片或者视频特别慢，你有遇到吗，我也是这个版本用了1年，最近折腾所有的ROM都有问题

najunuoyan, 2021-12-17:
> 

我这两天刚装上还没发现微信有问题，但是我发现打开苹果的APP store非常慢，周末我折腾一下DNS，看看能不能解决。

gili-gili, 2021-12-19:
R2S是从路由？

najunuoyan, 2021-12-19:
> R2S是从路由？

光猫->交换机->R2S

zzusec, 2021-12-19:
是网关 ---- 回复的原邮件 ***@***.***>发送日期2021年12月19日 16:08 ***@***.***> ***@***.******@***.***>主题Re: [klever1988/nanopi-openwrt] 【心态甭了】刷了12月所有的R2S的slim全都不分配ip地址 (Issue #976)

R2S是从路由？

[

{

"target": "https://github.com/klever1988/nanopi-openwrt/issues/976#issuecomment-997347394",

"url": "https://github.com/klever1988/nanopi-openwrt/issues/976#issuecomment-997347394",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"name": "GitHub",

"url": "https://github.com"

}

}

]

zzusec, 2021-12-19:
光猫- R2S-交换机- AP ---- 回复的原邮件 ***@***.***>发送日期2021年12月19日 19:33 ***@***.***> ***@***.******@***.***>主题Re: [klever1988/nanopi-openwrt] 【心态甭了】刷了12月所有的R2S的slim全都不分配ip地址 (Issue #976)

R2S是从路由？

光猫->交换机->R2S

[

{

"target": "https://github.com/klever1988/nanopi-openwrt/issues/976#issuecomment-997376207",

"url": "https://github.com/klever1988/nanopi-openwrt/issues/976#issuecomment-997376207",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"name": "GitHub",

"url": "https://github.com"

}

}

]

klever1988, 2021-12-19:
一般是光猫先接到软路由，再从软路由接到交换机啊

zzusec, 2021-12-19:
我的接发没问题，我帮好几个朋友都装过这个固件，他们都遇到微信打开图片和视频慢的问题，其他都正常，不知道是哪里出现问题了，我换别的固件，还是不如这个固件稳定 ---- 回复的原邮件 ***@***.***>发送日期2021年12月19日 22:16 ***@***.***> ***@***.******@***.***>主题Re: [klever1988/nanopi-openwrt] 【心态甭了】刷了12月所有的R2S的slim全都不分配ip地址 (Issue #976)

一般是光猫先接到软路由，再从软路由接到交换机啊

[

{

"target": "https://github.com/klever1988/nanopi-openwrt/issues/976#issuecomment-997399711",

"url": "https://github.com/klever1988/nanopi-openwrt/issues/976#issuecomment-997399711",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"name": "GitHub",

"url": "https://github.com"

}

}

]