stupidloud/nanopi-openwrt #977: R2S 插上8812cu  usb 无线网卡后不能启动系统

xiang5184, 2021-12-14:
大神您好, 我很喜欢你的slim固件, 可以自己加想要的插件.  但我发现最重要的一点就是当我插上 comfast 的8821cu  usb 网卡后就不能启动系统了, 无论多少次断电开机都不行 . 唯一的办法就是拔掉usb 网卡 才能开启系统.  
即使是开了系统后再插入网卡等一小会儿也会半死状态, 也不能后台登录了 .只有拔掉网卡断电再开机才能正常使用. 
我测试过使用骷颅头的固件可以正常使用这个comfast 8821cu usb无线网卡. 请问是否有办法解决这个问题? 谢谢!

klever1988, 2021-12-14:
差不多11月份中旬之后的代码存在bug 插入无线网卡之后就会死机 目前仍未解决

xiang5184, 2021-12-14:
你好，请问那个版本的才可以？我想用但不知哪个版本的才行，谢谢！

发自我的iPhone

klever1988, 2021-12-14:
11月之前
我的固件用的是天灵的源码，依赖组件什么的版本都比较新
如果你不追这种新，用骷髅头的固件也是可以的

xiang5184, 2021-12-14:
按你给的时间方向，今晚经过我不少于10次的刷机，终不负我, 锁定2021-10-28 是最后支持 8821cu usb wifi 不死机的版本。非常感谢大神！同时寄托后续的更新中得到解决给其它的网友。希望今晚测试的结果可以帮到其它的网友。再次感谢大神！

xiang5184, 2021-12-14:
我今晚按你给的方向刷了不少于10次固件，终于不负我努力，锁定2021-10-28 是最后支持 8812cu  USB wifi不死机的固件。感谢大神！希望后续的更新可以解决给其他网友，再次感谢！

发自我的iPhone

roxsky, 2021-12-16:
我也是同样的问题，用你反馈的10月28号的固件也解决了，谢谢了。

roxsky, 2021-12-16:
谢谢klever1988和xiang5184，我用的COMFAST 811AC无线网卡，和你有同样的问题，用你说的版本也可以正式使用了。

Deckard-f-Cain, 2021-12-25:
> 11月之前 我的固件用的是天灵的源码，依赖组件什么的版本都比较新 如果你不追这种新，用骷髅头的固件也是可以的

骷髅头的wifi驱动没有AC模式的，不太好用啊！但是 PASSWALL, SSPR+ 都很好用！

novicewu, 2022-04-24:
非常感谢！我折腾了两天，最终在这里找到了答案，并解决了问题。

dfldylan, 2022-05-05:
有同样的问题

q1ngyang, 2022-05-08:
> 

用了2021-10-28依然无法支持AC模式，只有legacy正常……