stupidloud/nanopi-openwrt #984: 关于Image builder

zhangguanzhang, 2021-12-19:
我在lede里用config make编译完成后
直接
```
make image PACKAGES=xxx
```
直接报错，我看readme里说的本项目支持image builder请问是如何支持的

klever1988, 2021-12-19:
看报错信息

zhangguanzhang, 2021-12-21:
```yaml
      - name: Build and deploy packages
        run: |
          ulimit -SHn 65000
          cd ~/lede
          while true; do make download -j && break || true; done
          make -j$[`nproc`+1] IGNORE_ERRORS=1
          mv `ls ~/lede/bin/targets/*/*/*imagebuilder*xz` ~/ib-$DEVICE.tar.xz
````
我看了下你这段，然后我在lede 的源码里 find grep 了下找到
```
$ find -type f -exec grep -i imagebuilder {} \;
Binary file ./.git/index matches
source "target/imagebuilder/Config.in"
#   Replacement for brcmImagebuilder
$(curdir)/builddirs:=linux sdk imagebuilder toolchain
$(curdir)/builddirs-install:=linux $(if $(CONFIG_SDK),sdk) $(if $(CONFIG_IB),imagebuilder) $(if $(CONFIG_MAKE_TOOLCHAIN),toolchain)
```
我开启了 imageBuilder 相关的:
```
CONFIG_IB_STANDALONE=y
CONFIG_IB=y
```
编译完后看到 `./bin/target/*/*` 下并没有 imageBuilder 目录生成：
```
guanzhang@guan:~/lede$ ll bin/targets/rockchip/armv8/
total 331124
drwxr-xr-x 3 guanzhang guanzhang       330 Dec 21 11:30 ./
drwxr-xr-x 3 guanzhang guanzhang        19 Dec 21 11:26 ../
-rw-r--r-- 1 guanzhang guanzhang     18201 Dec 21 11:26 config.buildinfo
-rw-r--r-- 1 guanzhang guanzhang       258 Dec 21 11:26 feeds.buildinfo
-rw-r--r-- 1 guanzhang guanzhang 192941463 Dec 21 11:30 openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-ext4-sysupgrade.img.gz
-rw-r--r-- 1 guanzhang guanzhang     19732 Dec 21 11:30 openwrt-rockchip-armv8-friendlyarm_nanopi-r2s.manifest
-rw-r--r-- 1 guanzhang guanzhang 146048872 Dec 21 11:29 openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-squashfs-sysupgrade.img.gz
drwxr-xr-x 2 guanzhang guanzhang     20480 Dec 21 11:30 packages/
-rw-r--r-- 1 guanzhang guanzhang       644 Dec 21 11:30 sha256sums
-rw-r--r-- 1 guanzhang guanzhang        16 Dec 21 11:26 version.buildinfo

```

zhangguanzhang, 2021-12-21:
@klever1988

zhangguanzhang, 2021-12-21:
> 编译结尾有没有输出 `make[3] -C target/imagebuilder install` 的字样？ 没有的话，直接 `make target/imagebuilder/compile V=sc` 看看输出啥呢

没有，然后我执行你的make后是下面输出
```
rstrip.sh: /home/guanzhang/lede/build_dir/target-aarch64_generic_musl/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/staging_dir/host/bin/.xxd.bin: shared object
rstrip.sh: /home/guanzhang/lede/build_dir/target-aarch64_generic_musl/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/staging_dir/host/bin/.xz.bin: shared object
rstrip.sh: /home/guanzhang/lede/build_dir/target-aarch64_generic_musl/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/staging_dir/host/bin/.xzdec.bin: shared object
rstrip.sh: /home/guanzhang/lede/build_dir/target-aarch64_generic_musl/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/staging_dir/host/bin/.zip.bin: shared object
rstrip.sh: /home/guanzhang/lede/build_dir/target-aarch64_generic_musl/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/staging_dir/host/bin/.zstd.bin: shared object
rstrip.sh: /home/guanzhang/lede/build_dir/target-aarch64_generic_musl/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/staging_dir/host/bin/.zyimage.bin: shared object
rstrip.sh: /home/guanzhang/lede/build_dir/target-aarch64_generic_musl/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/staging_dir/host/bin/.zytrx.bin: shared object
rstrip.sh: /home/guanzhang/lede/build_dir/target-aarch64_generic_musl/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/staging_dir/host/bin/.zyxbcm.bin: shared object
(cd /home/guanzhang/lede/build_dir/target-aarch64_generic_musl; \
	tar -I '/home/guanzhang/lede/staging_dir/host/bin/xz -7e -T0' -cf /home/guanzhang/lede/bin/targets/rockchip/armv8/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64.tar.xz openwrt-imagebuilder-rockchip-armv8.Linux-x86_64 \
	--mtime="Sat Dec 18 02:50:04 UTC 2021"; \
)
make[2]: Leaving directory '/home/guanzhang/lede/target/imagebuilder'
time: target/imagebuilder/compile#79.40#3.55#32.87
make[1]: Leaving directory '/home/guanzhang/lede'
```
然后输出目录有了 `openwrt-imagebuilder-rockchip-armv8.Linux-x86_64.tar.xz` 了，感谢大佬
```
$ ll bin/targets/rockchip/armv8/
total 365648
drwxr-xr-x 3 guanzhang guanzhang      4096 Dec 21 17:01 ./
drwxr-xr-x 3 guanzhang guanzhang        19 Dec 21 11:26 ../
-rw-r--r-- 1 guanzhang guanzhang     18201 Dec 21 11:26 config.buildinfo
-rw-r--r-- 1 guanzhang guanzhang       258 Dec 21 11:26 feeds.buildinfo
-rw-r--r-- 1 guanzhang guanzhang  35346900 Dec 21 17:01 openwrt-imagebuilder-rockchip-armv8.Linux-x86_64.tar.xz
-rw-r--r-- 1 guanzhang guanzhang 192941463 Dec 21 11:30 openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-ext4-sysupgrade.img.gz
-rw-r--r-- 1 guanzhang guanzhang     19732 Dec 21 11:30 openwrt-rockchip-armv8-friendlyarm_nanopi-r2s.manifest
-rw-r--r-- 1 guanzhang guanzhang 146048872 Dec 21 11:29 openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-squashfs-sysupgrade.img.gz
drwxr-xr-x 2 guanzhang guanzhang     20480 Dec 21 11:30 packages/
-rw-r--r-- 1 guanzhang guanzhang       644 Dec 21 11:30 sha256sums
-rw-r--r-- 1 guanzhang guanzhang        16 Dec 21 11:26 version.buildinfo

```

zhangguanzhang, 2021-12-21:
不过我还有点疑问，我看后面的阶段是利用 imageBuilder，那ci的 `Build r2s packages` 的意义在哪，里面的 ipk也打包进 `imageBuilder`了吗

zhangguanzhang, 2021-12-21:
@klever1988

klever1988, 2021-12-21:
对的 op的编译过程就是编译包然后生成rom
build packages就是编译包
make image就是生成rom
等于是拆成两步

zhangguanzhang, 2021-12-21:
> 对的 op的编译过程就是编译包然后生成rom build packages就是编译包 make image就是生成rom 等于是拆成两步

我下载了你的 `ib-device.tar.xz` ，解压后看到只有 `packages` 目录，而且我看里面有很多 ipk 文件:
```
$ cat repositories.conf
src imagebuilder file:packages
```
而我自己生成的 imageBuilder 里是是有从网上源获取，package 目录里只有少量的 ipk文件
```
## Place your custom repositories here, they must match the architecture and version.
# src/gz %n https://downloads.openwrt.org/snapshots
# src custom file:///usr/src/openwrt/bin/rockchip/packages

## Remote package repositories

## This is the local package repository, do not remove!
src imagebuilder file:packages

option check_signature
guanzhang@guan:~/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64$ cat repositories.conf.bak 
## Place your custom repositories here, they must match the architecture and version.
# src/gz %n https://downloads.openwrt.org/snapshots
# src custom file:///usr/src/openwrt/bin/rockchip/packages

## Remote package repositories
src/gz openwrt_core https://downloads.openwrt.org/snapshots/targets/rockchip/armv8/packages
src/gz openwrt_base https://downloads.openwrt.org/snapshots/packages/aarch64_generic/base
src/gz openwrt_luci https://downloads.openwrt.org/snapshots/packages/aarch64_generic/luci
src/gz openwrt_packages https://downloads.openwrt.org/snapshots/packages/aarch64_generic/packages
src/gz openwrt_routing https://downloads.openwrt.org/snapshots/packages/aarch64_generic/routing
src/gz openwrt_telephony https://downloads.openwrt.org/snapshots/packages/aarch64_generic/telephony

## This is the local package repository, do not remove!
src imagebuilder file:packages

option check_signature
```

请问下是哪里移动的 ipk 文件，我搜在你仓库里搜 `*.ipk` 没到任何信息

klever1988, 2021-12-21:
你生成ib的时候也是需要选中要编译的包，才会生成ipk的

zhangguanzhang, 2021-12-21:
好了，我忘记改.config 了，然后用你的那个 `make target/imagebuilder/compile` 这个相当于只开 IB=y 而没开 `CONFIG_IB_STANDALONE` ，开了俩后去搞就有了，感谢大佬解答

closed 2021-12-21 by zhangguanzhang

klever1988, 2021-12-21:
没事哈 有疑问尽管发

zhangguanzhang, 2021-12-22:
剩下一个就是关于cache的了，我看到有把整个源码目录打包作为cache，主要是哪个步骤是比较耗时的，哪些步骤可以保留目录来减少时间

klever1988, 2021-12-22:
https://forum.openwrt.org/t/using-precompiled-toolchain/87446

就只保存toolchain吧

zhangguanzhang, 2021-12-22:
toolchain不需要更新吗

klever1988, 2021-12-22:
源码有更新的话会自动重新编译的

zhangguanzhang, 2021-12-22:
```
root@guan:/home/guanzhang/lede# du -shx *
1.2G	bin
4.0K	BSDmakefile
25G	build_dir
68K	config
4.0K	Config.in
4.0K	COPYING
2.8G	dl
176K	doc
143M	feeds
4.0K	feeds.conf.default
488K	include
4.0K	key-build
4.0K	key-build.pub
4.0K	key-build.ucert
4.0K	key-build.ucert.revoke
56K	LICENSES
4.0K	Makefile
122M	package
4.0K	README
4.0K	README_EN.md
4.0K	README.md
16K	rules.mk
1.9M	scripts
3.9G	staging_dir
66M	target
119M	tmp
716K	toolchain
1.6M	tools
root@guan:/home/guanzhang/lede# ll build_dir/
total 28
drwxr-xr-x   6 guanzhang guanzhang   116 Dec 18 22:02 ./
drwxrwxr-x  19 guanzhang guanzhang  4096 Dec 21 20:54 ../
drwxr-xr-x  52 guanzhang guanzhang  4096 Dec 18 21:03 host/
drwxr-xr-x  25 guanzhang guanzhang  4096 Dec 21 20:55 hostpkg/
drwxr-xr-x 327 guanzhang guanzhang 12288 Dec 21 20:59 target-aarch64_generic_musl/
drwxr-xr-x  11 guanzhang guanzhang   243 Dec 18 21:43 toolchain-aarch64_generic_gcc-8.4.0_musl/
root@guan:/home/guanzhang/lede# du -shx build_dir/*
1.2G	build_dir/host
2.2G	build_dir/hostpkg
17G	build_dir/target-aarch64_generic_musl
5.2G	build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl
```
是 `build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl`这个目录吗

klever1988, 2021-12-22:
对

zhangguanzhang, 2021-12-27:
我看到文档 https://openwrt.io/docs/build-a-single-package/
编译所有包：
```
sed -i 's/# CONFIG_ALL .*/CONFIG_ALL=y/' .config
make defconfig
make package/compile -i -j V=99
```
为啥不这样搞，而是修改.config为m

klever1988, 2021-12-27:
> 我看到文档 https://openwrt.io/docs/build-a-single-package/ https://openwrt.org/zh-cn/doc/faq/before.installation#%E5%A6%82%E4%BD%95%E5%B0%86openwrt%E6%89%80%E6%9C%89packages%E5%9D%87%E7%BC%96%E8%AF%91%E8%BF%9B%E5%8E%BB 编译所有包：
> 
> ```
> sed -i 's/# CONFIG_ALL .*/CONFIG_ALL=y/' .config
> make defconfig
> make package/compile -i -j V=99
> ```
> 
> 为啥不这样搞，而是修改.config为m @klever1988

这样子要编译的东西太多了 很多用不到 而且github ci有6小时的限制肯定也跑不完

zhangguanzhang, 2022-01-08:
@klever1988 麻烦问下 `CONFIG_CCACHE=y` 是啥，之前搜了下ccache没看明白

klever1988, 2022-01-08:
就是启用ccache编译缓存

zhangguanzhang, 2022-01-08:
> 就是启用ccache编译缓存

这个有啥用

klever1988, 2022-01-08:
第二次编译可以节约40%-60%的时间 甚至更长

zhangguanzhang, 2022-01-09:
> 第二次编译可以节约40%-60%的时间 甚至更长

具体是缓存在哪个目录的，我现在缓存了下面俩目录快了俩小时，现在编译packages是4个半小时
```
            sudo mount --bind cache/build_dir    openwrt/build_dir
            sudo mount --bind cache/staging_dir  openwrt/staging_dir
```