stupidloud/nanopi-openwrt #988: SMB/CIFS 网络共享文件夹 挂载不上

lcl-101, 2021-12-25:
[image]
在挂载lu路径中没有
[image]

[image]

其他的openwrt系统可以

klever1988, 2021-12-25:
检查日志

lcl-101, 2021-12-28:
> 检查日志

[image]
只有这些

daiwei43211, 2021-12-28:
我也遇到了同样的问题！请问你有搞定吗？

daiwei43211, 2021-12-28:
我的邮箱：747376326@qq.com

klever1988, 2021-12-28:
cat /etc/init.d/cifs
会显示cifs插件的运行方式
其中
mount -t cifs -o vers=$smbver,user=$users,password=$pwd,iocharset=$iocharset,$agm //$server/$name $natpath
就是挂载命令
你可以匹配好参数，然后手动在终端挂载看看有什么出错提示

lcl-101, 2021-12-29:
> cat /etc/init.d/cifs 会显示cifs插件的运行方式 其中 mount -t cifs -o vers=$smbver,user=$users,password=$pwd,iocharset=$iocharset,$agm //$server/$name $natpath 就是挂载命令 你可以匹配好参数，然后手动在终端挂载看看有什么出错提示

[image]
我百度了 需要再命令前面加上 busybox 才能挂载上，不然就报错了，luc挂载哪里是不是需要调整

lcl-101, 2021-12-29:
cat /etc/init.d/cifs
执行完的结果

```bash
START=97

ENABLED=0
WORKGROUPD=0
IOCHARSET=0
GUEST=""
USERS=""
AGM=""

cifs_header() {
	local enabled
	local workgroup

	config_get enabled $1 enabled
	config_get workgroup $1 workgroup

	ENABLED=$enabled
	WORKGROUPD=$workgroup
}

mount_natshare() {
	local server
	local name
	local natpath
	local guest
	local users
	local pwd
	local agm
	local iocharset
	local smbver

	config_get server $1 server
	config_get name $1 name
	config_get natpath $1 natpath
	config_get guest $1 guest
	config_get users $1 users
	config_get pwd $1 pwd
	config_get agm $1 agm
	config_get iocharset $1 iocharset
	config_get smbver $1 smbver

	mkdir -p $natpath && chmod 777 $natpath
	#echo "mount -t cifs -o vers=$smbver,user=$users,password=$pwd,iocharset=$iocharset,$agm //$server/$name $natpath"
	mount -t cifs -o vers=$smbver,user=$users,password=$pwd,iocharset=$iocharset,$agm //$server/$name $natpath
}

start() {
	config_load cifs
	config_foreach cifs_header cifs

	if [ $ENABLED == 1 ]
	then {
		config_foreach mount_natshare natshare
		#echo "Cifs Mount succeed."
		}
	fi
}

stop() {
	mount | grep '//' | awk -F ' ' '{print$3}' | while read line; do
    #echo "umount -d -l $line"
    umount -d -l $line 2>/dev/null
	done
}
```

lcl-101, 2021-12-29:
哈哈哈，我改好了
[image]

把 /etc/init.d/cifs 文件中 mount 前加上 busybox 就可以了

[image]
已经测试能挂载上了