stupidloud/nanopi-openwrt #1004: 在线升级时报这个错“dd: invalid argument 'sparse' to 'conv'”怎么解决啊？

D4everT, 2022-01-13:
[image]

klever1988, 2022-01-13:
设备是？

liguilini, 2022-01-14:
我的R2S也碰到这个问题了，含docker版本

D4everT, 2022-01-14:
> 设备是？

R2S

klever1988, 2022-01-14:
这是因为调用到了不支持稀疏写入特性的busybox版dd命令，安装gnu dd命令可以解决，这个命令在较新的固件包是已安装的。
opkg update
opkg install coreutils-dd

之后再升级

D4everT, 2022-01-14:
root@ImmortalWrt:~# opkg update
Downloading https://mirrors.tencent.com/lede/snapshots/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_core
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_base
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_packages
Downloading https://mirrors.tencent.com/lede/releases/18.06.9/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_luci
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_routing
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_telephony
root@ImmortalWrt:~# opkg install coreutils-dd
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Installing coreutils-dd (8.32-6) to root...
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/packages/coreutils-dd_8.32-6_aarch64_generic.ipk
Collected errors:
 * wfopen: //usr/lib/opkg/info/coreutils-dd.control: Read-only file system.
 * wfopen: //usr/lib/opkg/info/coreutils-dd.postinst: Read-only file system.
 * wfopen: //usr/lib/opkg/info/coreutils-dd.prerm: Read-only file system.
 * wfopen: /usr/libexec/dd-coreutils: Read-only file system.
 * pkg_write_filelist: Failed to open //usr/lib/opkg/info/coreutils-dd.list: Read-only file system.
 * opkg_install_pkg: Failed to extract data files for coreutils-dd. Package debris may remain!
 * opkg_install_cmd: Cannot install package coreutils-dd.

安装coreutils-dd时也会出错

klever1988, 2022-01-14:
@D4everT 你的文件系统已经损坏，成只读无法写入的了。
拔卡重写固件或者e2fsck修复。

D4everT, 2022-01-15:
> @D4everT 你的文件系统已经损坏，成只读无法写入的了。 拔卡重写固件或者e2fsck修复。

谢谢大佬的回复。已经重新刷写了。