stupidloud/nanopi-openwrt #1011: 5.10内核Openclash守护程序不能运行

amwsuhje, 2022-01-17:
5.10内核Openclash守护程序不能运行，刷的是20220116版

mengxinz, 2022-01-17:
> 5.10内核Openclash守护程序不能运行，刷的是20220116版

lib/lua/luci/controller/openclash.lua第111行换成这样

return luci.sys.call("ps -w | grep openclash_watchdog.sh | grep -v grep >/dev/null") == 0
顺便关闭吧这个不是1988的问题

gekkjoe, 2022-01-17:
> > 5.10内核Openclash守护程序不能运行，刷的是20220116版
> 
> lib/lua/luci/controller/openclash.lua第111行换成这样
> 
> return luci.sys.call("ps -w | grep openclash_watchdog.sh | grep -v grep >/dev/null") == 0 顺便关闭吧这个不是1988的问题

[image]

老哥为啥我找不到这个路径

amwsuhje, 2022-01-18:
> > > 5.10内核Openclash守护程序不能运行，刷的是20220116版
> > 
> > 
> > lib/lua/luci/controller/openclash.lua第111行换成这样
> > return luci.sys.call("ps -w | grep openclash_watchdog.sh | grep -v grep >/dev/null") == 0 顺便关闭吧这个不是1988的问题
> 
> [image]
> 
> 老哥为啥我找不到这个路径

正确路径：/usr/lib/lua/luci/controller/openclash.lua

naolian64, 2022-01-23:
每次更新都要 手动更改么