stupidloud/nanopi-openwrt #1014: 请问最近几天的猫咪守护程序能正常运行了吗?

gekkjoe, 2022-01-19:
抱歉各位 有劳大哥们刷过的告知一下

weiyongzenqi, 2022-01-19:
试了,不行

yukkuna, 2022-01-20:
版本切到developer 升v.0.44.10就好了

klever1988, 2022-01-20:
猫咪是openclash？

mengxinz, 2022-01-20:
> 猫咪是openclash？

替换这个文件重启机器

/usr/lib/lua/luci/controller/openclash.lua第111行换成

return luci.sys.call("ps -w | grep openclash_watchdog.sh | grep -v grep >/dev/null") == 0

不加 -w 最大单行输出长度就是busybox默认是80个列宽（命令失败也是因为截断），除非设置了COLUMNS环境变量
https://github.com/mirror/busybox/blob/master/libbb/xfuncs.c#L265

所以解决方案有两个

设置 COLUMNS 环境变量
设置 busybox 的 ps 的 -w[w]选项 （一个w，列宽：132； 大于一个w，列宽：2*1024）参考：https://github.com/mirror/busybox/blob/master/procps/ps.c#L679

gekkjoe, 2022-01-20:
> 版本切到developer 升v.0.44.10就好了

请问怎么操作

gekkjoe, 2022-01-20:
> 替换这个文件重启机器
>
> /usr/lib/lua/luci/controller/openclash.lua第111行换成
>
> return luci.sys.call("ps -w | grep openclash_watchdog.sh | grep -v grep >/dev/null") == 0
>
> 不加 -w [...]
[image]

老哥 改了 重启了 还是一样 是不是我遗漏了什么

closed 2022-01-20 by gekkjoe