stupidloud/nanopi-openwrt #1021: 怎么安装cfdisk插件。我想扩容overlay

phfdick, 2022-01-26:
固件里没有cfdisk。
安装的时候更新软件源卡住不动。
nanopi r2s

phfdick, 2022-01-26:
系统-软件包-更新列表
正在执行软件包管理器
然后就卡住不动

选择“配置opkg”，出现如下错误提示
无法读取 /etc/opkg/distfeeds.conf：NoDataError: 未收到数据

phfdick, 2022-01-26:
尝试下载cfdisk的ipk。离线安装。
出现如下错误提示
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for cfdisk found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package cfdisk.
opkg install 命令失败，代码 255。

phfdick, 2022-01-26:
是23号的版本。设备是R2S。r2s-with-docker.img.gz。文件名是这个。

distfeeds.conf是啥。在命令行里执行这个么？

phfdick, 2022-01-27:
重新下载了26号的版本。问题解决。cfdisk可以安装上了。
但是在终端里输入cfdisk
返回cfdisk: cannot open /dev/sda:
找不到解决办法

phfdick, 2022-01-27:
cfdisk: cannot open /dev/sda: No such file or directory

klever1988, 2022-01-27:
你看的教程选择错误  sda是x86设备的驱动器抽象名称  r2s是mmc
你只是要扩容根分区的话 只需要原地升级一次我的固件就自动扩容了啊何必折腾