stupidloud/nanopi-openwrt #1026: R2s 2022-02-02 11:52:00 ERROR 404: NOT FOUND.

APPX-Studio, 2022-02-02:
[image]
releases/download/2022-02-02/r2.img.gz
Resolving ghproxy.com... 146.56.43.67
Connecting to ghproxy.com|146.56.43.67|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2022-02-02 11:48:16 ERROR 404: NOT FOUND.

gzip: stdin: unexpected end of file
awk: md5sum.txt: No such file or directory
root@BLACK:/# 
root@BLACK:/# wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | ver=-slim bash
Downloading https://mirrors.cloud.tencent.com/lede/releases/19.07.5/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://mirrors.cloud.tencent.com/lede/releases/19.07.5/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://mirrors.cloud.tencent.com/lede/releases/19.07.5/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://mirrors.cloud.tencent.com/lede/releases/19.07.5/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Package coreutils-truncate (8.30-2) installed in root is up to date.
准备更新到2022-02-02
--2022-02-02 11:52:00--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2022-02-02/r2-slim.img.gz
Resolving ghproxy.com... 146.56.158.111
Connecting to ghproxy.com|146.56.158.111|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2022-02-02 11:52:00 ERROR 404: NOT FOUND.

mengxinz, 2022-02-02:
> [image] releases/download/2022-02-02/r2.img.gz Resolving ghproxy.com... 146.56.43.67 Connecting to ghproxy.com|146.56.43.67|:443... connected. HTTP request sent, awaiting response... 404 NOT FOUND [...]

r2s今天的包还没编译出来

lwl19, 2022-02-07:
包地址错了吧 (脚本里面)r2.img.gz --->  r2s.img.gz(实际地址)  少了个字母，404

klever1988, 2022-02-07:
在终端执行
cat /etc/board.json
看看

lwl19, 2022-02-07:
cat /etc/board.json:

{
        "model": {
                "id": "friendlyelec,nanopi-r2",
                "name": "FriendlyElec NanoPi R2S"
        },
        "network": {
                "lan": {
                        "ifname": "eth1",
                        "protocol": "static"
                },
                "wan": {
                        "ifname": "eth0",
                        "protocol": "dhcp"
                }
        }
}

lwl19, 2022-02-07:
id 要修改为r2s吧

lwl19, 2022-02-07:
直接修改id更新后，挂掉了。重新刷机2022/2/7的版本，发现v2ray协议不能用