stupidloud/nanopi-openwrt #1028: R4S在线升级失败

huxing188, 2022-02-03:
> root@HuWrt:~# wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | bash
Downloading https://mirrors.tencent.com/lede/snapshots/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_core
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_base
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_packages
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_luci
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_routing
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_telephony
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Package coreutils-truncate (8.32-6) installed in root is up to date.
准备更新到2022-02-02
--2022-02-03 13:24:53--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2022-02-02/r4s.img.gz
Resolving ghproxy.com... 146.56.158.111
Connecting to ghproxy.com|146.56.158.111|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2022-02-03 13:24:53 ERROR 404: NOT FOUND.

huxing188, 2022-02-03:
在线升级一直提示这样，并且使用在线升级后，DOCKER页面就不见是啥情况。
还有其他升级方式吗？（不抹除现有的一些数据信息）

chenyuyou, 2022-02-03:
新版本的没编译，等新版本编译成功

hardass, 2022-02-07:
> 在线升级一直提示这样，并且使用在线升级后，DOCKER页面就不见是啥情况。
> 还有其他升级方式吗？（不抹除现有的一些数据信息）

关于在线升级docker不见了的问题，之前有人问过同样的问题，脚本记得要加上参数 with-docker，就可以了
wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | ver=-with-docker bash

lcl-101, 2022-03-08:
> > 在线升级一直提示这样，并且使用在线升级后，DOCKER页面就不见是啥情况。
> > 还有其他升级方式吗？（不抹除现有的一些数据信息）
> 
> 关于在线升级docker不见了的问题，之前有人问过同样的问题，脚本记得要加上参数 with-docker，就可以了 wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | ver=-with-docker bash

slim版本的保留docker应该怎么写了