stupidloud/nanopi-openwrt #1031: 安装app出错

zhdw22114632, 2022-02-05:
[image]

klever1988, 2022-02-05:
空间剩余不足啊 那不是说着吗

zhdw22114632, 2022-02-06:
使用在线升级不是自动扩容空间么？怎么还是原来的空间大小呀

zhdw22114632, 2022-02-06:
> 使用在线升级不是自动扩容空间么？怎么还是原来的空间大小呀

klever1988, 2022-02-06:
看一下你的“剩余空间”图例

zhdw22114632, 2022-02-06:
> 看一下你的“剩余空间”图例
第一次用这个slim版本，想自己装一下passwall，发现是空间小不能扩容，还发现这几天的版本不能用京东签到，我再研究研究
[image]
[image]

klever1988, 2022-02-06:
看起来你只是刷过一次固件 没有在线升级过啊

zhdw22114632, 2022-02-06:
> 看起来你只是刷过一次固件 没有在线升级过啊

刷完系统以后没有科学，在线升级没反应，我明天前面弄个能科学的升一下试试