stupidloud/nanopi-openwrt #1032: 请问点击passwall手动更新组件以后 无法使用v2ray了

Sev73n, 2022-02-07:
[image]
检测有5.0的新版本 就点了一下 后台跑的很顺畅 还有什么移动中的提示 显示更新成功以后
就不能用了 估计是更新包位置检测不到了之类的
有没有办法重置回以前的版本？ 只能重新刷机吗

Sev73n, 2022-02-07:
终端里运行提示 A unified platform for anti-censorship.
[image]

klever1988, 2022-02-07:
这个没有深入去看也不懂具体原因为何， 疑难杂症还是升级重刷固件解决吧，比较快

lwl19, 2022-02-07:
我重新刷了今天的固件,v2ray确实不能用了

lwl19, 2022-02-08:
v2ray组件的客服端和服务端的版本最好一致，会出现不兼容的问题。升级服务端可解决。

kinboyw, 2022-03-05:
> [image] 检测有5.0的新版本 就点了一下 后台跑的很顺畅 还有什么移动中的提示 显示更新成功以后 就不能用了 估计是更新包位置检测不到了之类的 有没有办法重置回以前的版本？ 只能重新刷机吗

https://www.geekersq.cc/2022/01/passwall-fix-v2ray/
同样的问题，查了下，应该是版本问题，v2ray 5.0以上版本目前都还是开发版， passwall目前还不支持开发版，恢复到最新稳定版4.44.0可以解决

cunz1974, 2022-11-26:
我是恢复出厂设置后再重新配置的。恢复后就会到以前的可用版本了。