stupidloud/nanopi-openwrt #1048: dockerd无法安装，内核不兼容

EcoRIIHC, 2022-02-20:
安装dockerd报错了，请问如何解决？ 

```
Installing dockerd (20.10.12-1) to root...
Downloading https://mirrors.ustc.edu.cn/openwrt/releases/21.02.0/packages/aarch64_generic/packages/dockerd_20.10.12-1_aarch64_generic.ipk
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-crypto-hash
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-crypto-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-lib-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-crypto-acompress
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-lib-lzo
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-lib-zlib-inflate
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-lib-zlib-deflate
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-lib-raid6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-lib-xor
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-lib-zstd
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-fs-btrfs
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-crypto-null
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-crypto-aead
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-crypto-pcompress
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-pcompress found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-crypto-manager
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-dax
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-dax found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-dm
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-dm found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-ipt-extra
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-br-netfilter
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-br-netfilter found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-ikconfig
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ikconfig found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-nfnetlink
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-nf-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-ipt-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-nf-conntrack-netlink
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-conntrack-netlink found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-nf-ipvs
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-ipvs found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-nf-nat
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1) for kmod-veth
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-veth found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for dockerd:
 * 	kernel (= 5.4.143-1-fb881fbbae69f30da18e7c6eb01310c1)
 * opkg_install_cmd: Cannot install package dockerd.

```

closed 2022-02-23 by EcoRIIHC

peoplechinapower, 2022-05-25:
解决了吗

klever1988, 2022-05-25:
已经提供了docker版本，也可以从slim的内置源安装docker，不明白为什么要从官方源安装docker。
内核不兼容是普遍存在的，每次编译固件都会产生不一样的内核哈希值，这就是为什么slim要内置一个完全匹配的本地软件源
想要内核匹配官方软件源，要么下载官方编译的固件，或者修改内核的哈希值，别无他法

GreenTeodoro839, 2022-07-18:
> 已经提供了docker版本，也可以从slim的内置源安装docker，不明白为什么要从官方源安装docker。 内核不兼容是普遍存在的，每次编译固件都会产生不一样的内核哈希值，这就是为什么slim要内置一个完全匹配的本地软件源 想要内核匹配官方软件源，要么下载官方编译的固件，或者修改内核的哈希值，别无他法

能否做一个自己改在线源+本地kmod源的包，或者是提供编译产生的kmod ipk安装包