stupidloud/nanopi-openwrt #1049: 0219版本x86固件，使用openclash无法联网

WilsonXu0512, 2022-02-20:
虚拟机跑的x86，使用默认版固件（x86.img）
之前跑的21年10月份固件，内核5.4，一直可以正常用
今天刷20220219版（重装，非在线升级），内核5.10，所有设置都没变，openclash可以正常启动但是死活无法连上网，内外都不行。
设置一个都没改，试过老配置备份——刷机后上传恢复，也试过重刷后一点点手动设置，都无法搞定
只用openclash一个插件，没有其他干扰。x86软路由做旁路由模式，网关之类设置也都没改。怀疑是不是防火墙的问题？但是水平有限又不会自己改

dxg-aceinna, 2022-02-21:
可以参考这个issue，目前还没看到正式的解决方法。
#1040

WilsonXu0512, 2022-02-21:
> 可以参考这个issue，目前还没看到正式的解决方法。 #1040

发issue之前就看到了，水平不够看了两遍没看懂……而且，似乎问题也没有解决吧

dxg-aceinna, 2022-02-21:
是的，截止到上周五，还未解决，并且issue里面也没有进一步的讨论。然后我就暂时先换用别的固件了。

Lyt99, 2022-02-25:
我也是上周刷的版本挂了，看起来是iptables默认链接到了nf_tables的问题，手动换成legacy解决了，不知道适不适用
ssh 登录到路由器，执行
```bash
rm $(which iptables)
ln -s /usr/sbin/iptables $(which iptables-legacy)
```
重新启动openclash，刷新iptables规则就好了

L1cardo, 2022-02-28:
我被这个问题困扰了四五天！！ 终于解决了，感谢感谢！！

上面的评论有点乱，我来总结一下
运行这两个命令
```bash
rm $(which iptables)
iptables
```

这时候如果显示如下则说明成功，重点是`(legacy)`，有这个字样就行
```bash
iptables v1.8.7 (legacy): no command specified
Try `iptables -h' or 'iptables --help' for more information.
```

然后关闭opanclash，然后重新打开openclash

如上操作应该就能解决

WilsonXu0512, 2022-02-28:
网友们热心帮大佬修复上游留下的bug，真心令人感动

1098880928aa, 2022-03-03:
root@ImmortalWrt:/etc/crontabs# **ln -sf $(which iptables-legacy) /usr/sbin/iptables**
ln: /usr/sbin/iptables: File exists
尝试直接创建软链接失败

root@ImmortalWrt:/etc/crontabs# **ll /usr/sbin/iptables**
lrwxrwxrwx    1 root     root            27 Feb 19 21:57 /usr/sbin/iptables -> /usr/sbin/xtables-nft-multi*
root@ImmortalWrt:/etc/crontabs# **rm /usr/sbin/iptables**
rm: can't remove '/usr/sbin/iptables': Read-only file system
尝试删除iptables，只读系统也无法删除

root@ImmortalWrt:/etc/crontabs# **mount -o remount -rw  /dev/root**
mount: mounting /dev/root on / failed: Read-only file system
尝试挂载读写，也失败了

这该怎么办？