stupidloud/nanopi-openwrt #1075: 2022/4/2更新后passwall和openclash都没了，而且ssrplus+还无法订阅

wanghuzhou, 2022-04-02:
暂时回退到2022/4/1了

InsOnYou, 2022-04-03:
安装OpenClash时，kmod依赖提示内核不对，导致安装失败。

Biyanbin, 2022-04-03:
我甚至连安装都没办法 有别的版本可以先凑合用的吗？

L1cardo, 2022-04-03:
最新版openclash需要安装kmod-inet-diag依赖
参考https://github.com/vernesong/OpenClash/issues/2215

ppyohhh, 2022-04-03:
> 最新版openclash需要安装kmod-inet-diag依赖 参考[vernesong/OpenClash#2215](https://github.com/vernesong/OpenClash/issues/2215)

今天发布的43版本 重新装一下把

wanghuzhou, 2022-04-03:
> 最新版openclash需要安装kmod-inet-diag依赖 参考[vernesong/OpenClash#2215](https://github.com/vernesong/OpenClash/issues/2215)

感谢大佬

klever1988, 2022-04-03:
一个失误，导致opclash没有正确编译，现已修复~

younasiqw, 2022-04-04:
4月版本没有smartdns，安装了luci 和 smartdns也不显示选项有大佬怎么解决吗

closed 2022-04-04 by wanghuzhou

klever1988, 2022-04-05:
passwall也已加回，帮忙测试一下正常不 @wanghuzhou

wanghuzhou, 2022-04-05:
[image]
[image]
已正常使用，就是订阅页面有点ui问题，但不影响输入

本次升级我是通过下载img.gz上传 保留配置升级的，openclash无法使用，看日志发现Country.mmdb好像出问题了，删了之后重新启动自动下载新的就没问题了。
@klever1988