stupidloud/nanopi-openwrt #1076: 求大佬在R2S普通版本中加入AdguardHome、Passwall和SmartDNS

WilsonXu0512, 2022-04-04:
求大佬在R2S普通版本中加入AdguardHome、Passwall和SmartDNS

自己安装软件包，AdguardHome总是存在设置无法保存的问题
SmartDNS装了Luci还是无法在服务无法显示
Paswall本来是做openclash备用的，现在也没了

求大佬把常用服务的插件都编译进去，和以前的Immortal项目一样就好
Immortal项目用一年用习惯了，现在OpenWrt的项目插件太少，又要重新配置好多东西

younasiqw, 2022-04-04:
SmartDNS 选择[luci-app-smartdns.1.2022.03.29-1920.all-luci-compat-all.ipk](https://github.com/pymumu/smartdns/releases/download/Release36/luci-app-smartdns.1.2022.03.29-1920.all-luci-compat-all.ipk)
安装compat版的就可以显示了不过4月版本有bug SmartDNS总是会自动关闭 不知道是不是我单人问题

ws4125570, 2022-04-04:
我还是老老实实用2月的版本吧，楼主提的这三个插件有两个是我一直在用的，没了等于废了

Que-God, 2022-04-05:
ssr+也可以用其实

klever1988, 2022-04-05:
AdguardHome太久没有更新了，其他两个插件已添加，要测试一下。。。

ws4125570, 2022-04-05:
> AdguardHome太久没有更新了，其他两个插件已添加，要测试一下。。。

adh我记得上个月有更新过哦

klever1988, 2022-04-05:
> 

哪个仓库地址发一下

klever1988, 2022-04-05:
> > AdguardHome太久没有更新了，其他两个插件已添加，要测试一下。。。
> 
> adh我记得上个月有更新过哦

哪个仓库地址发一下

ws4125570, 2022-04-05:
> > > AdguardHome太久没有更新了，其他两个插件已添加，要测试一下。。。
> > 
> > 
> > adh我记得上个月有更新过哦
> 
> 哪个仓库地址发一下

插件可能没更新的，我说的是核心版本上个月有更新，最新核心版本是0.107.5。如果不是一个东西的话，不好意思哦，搞混了

WilsonXu0512, 2022-04-05:
> AdguardHome太久没有更新了，其他两个插件已添加，要测试一下。。。

> > > AdguardHome太久没有更新了，其他两个插件已添加，要测试一下。。。
> > 
> > 
> > adh我记得上个月有更新过哦
> 
> 哪个仓库地址发一下

大佬你好，
AdguardHome官方的仓库地址是：https://github.com/AdguardTeam/AdGuardHome
luci-app-adguardhome的仓库地址是：https://github.com/rufengsuixing/luci-app-adguardhome

AdguardHome内核一直有更新，最新版本v0.107.5，2022年3月4号更新的
太久没有更新是Luci-app(更新到2019年12月)，版本1.8-9，不影响使用的，原来的ImmortalWrt用的也是1.8-9的Luci,
这个软件和clash类似，重要的是经常更新规则和内核，Luci包的更新不频繁

再次感谢添加了另外两个插件，稍后测试