stupidloud/nanopi-openwrt #1090: 目前对我来说，固件基本完美！就差adguardhome和mosdns了！

liuyuan29, 2022-04-12:
adguadhome添加不进固件，mosdns没办法单独启动，而且跟ssr+联动也是没启动，不知怎么回事？

h6fYe34KFSiH, 2022-04-14:
adguardhome 装到 docker 里，目前用着没毛病

liuyuan29, 2022-04-15:
> adguadhome暂时可以用ssh装进去： curl -s -S -L https://raw.githubusercontent.com/AdguardTeam/AdGuardHome/master/scripts/install.sh | sh -s -- -v 然后安装Luci包：luci-app-adguardhome_1.8-11_all.ipk，下载地址：https://github.com/rufengsuixing/luci-app-adguardhome/releases ，上传安装，重启
> 
> 重启之后AdguardHome的菜单应该能出来，但是现在还不能用，要改以下东西： 执行文件路径：/opt/AdGuardHome/AdGuardHome 配置文件路径：/opt/AdGuardHome/AdGuardHome.yaml 工作目录：/opt/AdGuardHome 工作目录备份路径：/opt/AdGuardHome
> 
> 改完后再重启，就可以进行配置了。
> 
> 如果你有老的配置存档，在这一步把老配置的.yaml文件用WinSCP上传覆盖/opt/AdGuardHome/AdGuardHome.yaml这个文件，就好了
> 
> 步骤有点多，但确保稳妥能用

试过了，菜单出不来！

liuyuan29, 2022-04-15:
mosdns也是没菜单，跟ssr+联动也是启动不了的！

klever1988, 2022-04-24:
@liuyuan29 mosdns已经修复了，缺geo库而已