stupidloud/nanopi-openwrt #1091: r2s在线升级后docker的容器和已下载的镜像没了

kid-oh, 2022-04-12:
不知道是不是默认就是会把docker配置初始化，在线升级选的是docker版本，升级完进去发现容器都没了

ni7ht, 2022-04-13:
能问下在线怎么升级docker版吗？主页没用docker版升级指令，我是下载下来本地上传刷完docker也容器镜像也都没了，而且2022-04-11版passwall失效了，无法科学上网。

kid-oh, 2022-04-13:
> 能问下在线怎么升级docker版吗？主页没用docker版升级指令，我是下载下来本地上传刷完docker也容器镜像也都没了，而且2022-04-11版passwall失效了，无法科学上网。

wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | ver=-with-docker bash

kid-oh, 2022-04-13:
> 能问下在线怎么升级docker版吗？主页没用docker版升级指令，我是下载下来本地上传刷完docker也容器镜像也都没了，而且2022-04-11版passwall失效了，无法科学上网。

我后面重装了普通版本不带docker passwall目前没遇到问题

ni7ht, 2022-04-14:
> > 能问下在线怎么升级docker版吗？主页没用docker版升级指令，我是下载下来本地上传刷完docker也容器镜像也都没了，而且2022-04-11版passwall失效了，无法科学上网。
> 
> wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | ver=-with-docker bash

感谢，我退到上一个版本了

ni7ht, 2022-04-14:
> > > > 能问下在线怎么升级docker版吗？主页没用docker版升级指令，我是下载下来本地上传刷完docker也容器镜像也都没了，而且2022-04-11版passwall失效了，无法科学上网。
> > > 
> > > 
> > > wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | ver=-with-docker bash
> > 
> > 
> > 感谢，我退到上一个版本了
> 
> 上一个版本不会掉容器吗

重新拉了，先用着2022-04-07