stupidloud/nanopi-openwrt #1092: 最新版ssr+mosdns动态模式开启后所有网站都无法上网了

Agoni007, 2022-04-14:
节点是vemss的v2ray节点，使用默认的mosdns模式，任何网站都打不开了，改成其他的模式都是正常的，其他的没有问题。

TechMechannic, 2022-04-16:
我也用不了，感觉monsdns模式很好用。就是时间长了容易不稳定

klever1988, 2022-04-23:
昨天有修复，再试试。

TechMechannic, 2022-04-24:
> 昨天有修复，再试试。

已试，还是不行。但是关闭防火墙就能用了

TechMechannic, 2022-04-24:
> 对 一直不知道什么原因 ssrp状态里全部连不通 4.11和4.20版本都不行 之前的版本没试过就不知道了 如果有解决办法的大佬麻烦踢我一下 [image]

关闭防火墙暂时能用

kid-oh, 2022-04-24:
> > 对 一直不知道什么原因 ssrp状态里全部连不通 4.11和4.20版本都不行 之前的版本没试过就不知道了 如果有解决办法的大佬麻烦踢我一下 [image]
> 
> 关闭防火墙暂时能用

我更新新版没关防火墙可以用

TechMechannic, 2022-04-24:
> > > 对 一直不知道什么原因 ssrp状态里全部连不通 4.11和4.20版本都不行 之前的版本没试过就不知道了 如果有解决办法的大佬麻烦踢我一下 [image]
> > 
> > 
> > 关闭防火墙暂时能用
> 
> 我更新新版没关防火墙可以用

能用就好了 我是x86的