stupidloud/nanopi-openwrt #1106: 升级脚本无法获取R2S对应的img

poorest1212, 2022-04-22:
root@DJ-R2S:/# wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | ver=-slim bash
Downloading https://nanopi-r2s-opkg-feeds.songchenwen.com/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/songchenwen
Package coreutils-truncate (9.0-2) installed in root is up to date.
准备更新到2022-04-21
--2022-04-22 12:50:24--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2022-04-21/r2-slim.img.gz <----问题是不是在脚本获取model id的字符串处理？
Resolving ghproxy.com... 138.2.78.227
Connecting to ghproxy.com|138.2.78.227|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2022-04-22 12:50:27 ERROR 404: NOT FOUND.

klever1988, 2022-04-22:
修改了升级脚本 再试一下吧

poorest1212, 2022-04-23:
好了～正则太复杂看不懂哈

closed 2022-04-23 by poorest1212