stupidloud/nanopi-openwrt #1112: 新版的IPV6又搞不定了。

ws4125570, 2022-04-25:
2月之前的版本ipv6还行，可以正常使用。4月切换了核心之后，接口设置和内容都不一样了，往上也搜了一堆设置方法，不管咋整还是连不上ipv6。
有没有高手指点一下，新版怎么设置？

klever1988, 2022-04-25:
[image]

我昨晚刚升级，没有问题啊

amwsuhje, 2022-04-25:
> [image]
> 
> 我昨晚刚升级，没有问题啊

 怎么我刚在线升级成功，版本还是显示4-23的

ws4125570, 2022-04-25:
> 

请问大大就是默认设置就可以连通截图显示的ipv6吗？后面设备ipv6 使用也没问题的喽？
见了鬼了:(

ws4125570, 2022-04-25:
> [image]
> 
> 我昨晚刚升级，没有问题啊

请问大大就是默认设置就可以连通截图显示的ipv6吗？后面设备ipv6 使用也没问题的喽？
见了鬼了:(

klever1988, 2022-04-25:
@amwsuhje 那就再升级一次，

@ws4125570 是的，初始状态下设定好拨号就可连通v6了啊。我还是复原之前immortalwrt的固件设定档来的。

ws4125570, 2022-04-25:
@klever1988 那接口设置wan6和lan的IPV6设置是否可以借鉴下。我是光猫拨号，r2s做软路由，之前immortalwrt是可以实现连通ipv6，但是现在就不行了，有点郁闷。

amwsuhje, 2022-04-25:
> @amwsuhje 那就再升级一次，
> 
> @ws4125570 是的，初始状态下设定好拨号就可连通v6了啊。我还是复原之前immortalwrt的固件设定档来的。

刚刚重新卡刷了一次，进去看还是显示4-23，之前试过在线脚本升，把固件包上传到/tmp目录用命令升，都是显示4-23

ws4125570, 2022-04-25:
@amwsuhje 我刷了今天的包也是显示4.23，可能是个版本的bug吧

amwsuhje, 2022-04-25:
> @amwsuhje 我刷了今天的包也是显示4.23，可能是个版本的bug吧

我昨天在线更的4-24版本也是显示4-23，今天更的4-25也是显示4-23，有点怀疑是不是没更成功。直到刚刚看到卡刷也是4-23

klever1988, 2022-04-25:
不是bug 昨天编译25的确实没有成功（只有slim成功了） 固件是使用23号的二进制包生成的所以23

younasiqw, 2022-04-30:
> > @amwsuhje 那就再升级一次，
> > @ws4125570 是的，初始状态下设定好拨号就可连通v6了啊。我还是复原之前immortalwrt的固件设定档来的。
> 
> 刚刚重新卡刷了一次，进去看还是显示4-23，之前试过在线脚本升，把固件包上传到/tmp目录用命令升，都是显示4-23

有没有一种可能 我是说可能 你游览器缓存需要清理一下后重新查看 :))