stupidloud/nanopi-openwrt #1114: 能否给mosdns换一个带菜单的啊？ssrp带动不起来，运行不了！

liuyuan29, 2022-04-28:
能否给mosdns换一个带菜单的啊？ssrp带动不起来，运行不了！

klever1988, 2022-04-28:
进入终端，执行
`/usr/bin/mosdns -dir /etc/mosdns/ -c ./config.yaml`
看看回显什么

liuyuan29, 2022-05-01:
原来是我的机场节点问题，搞得我以为不行！
2022-05-01T08:16:43.613Z        info    v3/main.go:157  current working directory: /etc/mosdns/
2022-05-01T08:16:43.613Z        info    v3/main.go:223  mosdns ver: v3.1.1
2022-05-01T08:16:43.613Z        info    v3/main.go:224  arch: amd64, os: linux, go: go1.18.1
2022-05-01T08:16:43.613Z        info    coremain/run.go:67      loading config  {"file": "./config.yaml"}
2022-05-01T08:16:43.619Z        fatal   main_server     server/server.go:129    server exited unexpectedly      {"error": "listen udp 0.0.0.0:5335: bind: address already in use"}