stupidloud/nanopi-openwrt #1120: fork后自己编译无法发布固件是怎么回事啊

549751001, 2022-05-04:
Run svenstaro/upload-release-action@v[2](https://github.com/549751001/nanopi-openwrt/runs/6287502337?check_suite_focus=true#step:4:2)
  with:
    repo_token: ***
    file: ./release/*
    tag: 2022-05-0[4](https://github.com/549751001/nanopi-openwrt/runs/6287502337?check_suite_focus=true#step:4:4)
    file_glob: true
    overwrite: true
    release_name: 2022-0[5](https://github.com/549751001/nanopi-openwrt/runs/6287502337?check_suite_focus=true#step:4:5)-04 自动发布
  env:
    DEVICE: r4s
    strDate: 2022-05-04
    strDevice: R4S
Error: Resource not accessible by integration
提示的错误

Xuchengen, 2022-05-05:
我也遇到这样的问题

Xuchengen, 2022-05-06:
> 同样的问题，请问您是怎么解决的

代码仓库管理页面中，配置几个选项就好了。

troyliu0105, 2022-05-07:
> 代码仓库管理页面中，配置几个选项就好了。

谢谢🙏，搞定了。

GeeKaven, 2022-05-10:
> > 代码仓库管理页面中，配置几个选项就好了。
> 
> 谢谢🙏，搞定了。

能否问下如何搞定的？

Mars-Sea, 2023-03-03:
> > > 代码仓库管理页面中，配置几个选项就好了。
> > 
> > 
> > 谢谢🙏，搞定了。
> 
> 能否问下如何搞定的？

在setting -> Action -> general的最下面Workflow permissions里，把Read and write permissions打勾和Allow GitHub Actions to create and approve pull requests勾上保存就可以了