stupidloud/nanopi-openwrt #1123: R2C全新刷入5.8的固件拨号后，内核日志一直在刷 test 512

lfdfxl, 2022-05-08:
R2C全新刷入5.8的固件拨号后，内核日志一直在刷 test 512，但是不影响正常使用，大佬有时间帮忙看看吧。

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.113 (runner@fv-az198-826) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r4386-7a7f9d647) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP PREEMPT Sat May 7 18:27:37 2022
[    0.000000] Machine model: FriendlyElec NanoPi R2C
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] On node 0 totalpages: 261632
[    0.000000]   DMA zone: 4088 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 261632 pages, LIFO batch:63
[    0.000000] cma: Reserved 8 MiB at 0x000000003f400000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 21 pages/cpu s48152 r8192 d29672 u86016
[    0.000000] pcpu-alloc: s48152 r8192 d29672 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1001996K/1046528K available (8382K kernel code, 2378K rwdata, 2392K rodata, 1728K init, 745K bss, 36340K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x284/0x3c4 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000009] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001528] Console: colour dummy device 80x25
[    0.001995] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.002945] pid_max: default: 32768 minimum: 301
[    0.003657] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004346] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.007697] rcu: Hierarchical SRCU implementation.
[    0.008389] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.010129] smp: Bringing up secondary CPUs ...
[    0.011283] Detected VIPT I-cache on CPU1
[    0.011392] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.012249] Detected VIPT I-cache on CPU2
[    0.012330] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.013144] Detected VIPT I-cache on CPU3
[    0.013219] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.013369] smp: Brought up 1 node, 4 CPUs
[    0.016596] SMP: Total of 4 processors activated.
[    0.017033] CPU features: detected: 32-bit EL0 Support
[    0.017597] CPU features: detected: CRC32 instructions
[    0.018419] CPU: All CPU(s) started at EL2
[    0.018821] alternatives: patching kernel code
[... 977 lines cut ...]
[  344.317900] test 512
[  345.043249] test 512
[  345.276687] test 512
[  345.284026] test 512
[  345.301849] test 512
[  345.784284] test 512
[  346.096957] test 512
[  346.193682] test 512
[  348.810299] test 512
[  349.030207] test 512
[  349.387956] test 512
[  351.423548] test 512
[  354.982293] test 512
[  355.024960] test 512
[  355.028126] test 512
[  355.796575] test 512
[  356.849204] test 512
[  357.029998] test 512
[  359.026599] test 512
[  360.235385] test 512
[  360.270702] test 512
[  362.061277] test 512
[  362.071595] test 512
[  362.589579] test 512
[  363.057854] test 512

klever1988, 2022-05-08:
之前的版本会这样子吗
把tcp加速关闭了试试

lfdfxl, 2022-05-08:
> 之前的版本会这样子吗 把tcp加速关闭了试试

自从R2C网卡拨号问题解决号，之后的版本都存在这个问题。
刚才关闭TTurbo ACC的软件流量分载后不再刷这个日志。

klever1988, 2022-05-08:
那个加速没有实际效果，不要开了

lam2888, 2022-05-08:
r2s with docker 0508版本在线升级中，出现以下提示，然后就不动了。把内存卡插在读卡器上，sudo fsck.ext4 -y /dev/sdb1和sdb2,一定是要自动修复的。
root@OpenWrt:~# tar: write: Broken pipe

lfdfxl, 2022-05-08:
> 那个加速没有实际效果，不要开了

好的，感谢大佬！

closed 2022-05-08 by lfdfxl