stupidloud/nanopi-openwrt #1126: r2s最近版本总是重启

L1cardo, 2022-05-09:
不知道有没有朋友遇到过，最近一周的版本总是重启，然后重启完了系统分区变成只读了，啥都改不了，文件也不能上传
1. 起初我怀疑是供电不足，换了5v3a的充电器，还是不行
2. 我怀疑是不是openclash的问题，但是不操作openclash，在其他页面也会重启
3. 难道是tf卡的问题？

有没有朋友有什么意见

klever1988, 2022-05-09:
用读卡器重新刷一次固件试试

yuanzhuohao, 2022-05-09:
@klever1988  我用r2s也差不多有这个 ，我用tf轮流重刷也一样

klever1988, 2022-05-09:
> @klever1988 我用r2s也差不多有这个问题 ，我用tf轮流重刷也一样

cpufreq插件把最高频降低一个阶梯试试看

yuanzhuohao, 2022-05-10:
> > @klever1988 我用r2s也差不多有这个问题 ，我用tf轮流重刷也一样
> 
> cpufreq插件把最高频降低一个阶梯试试看

我用slim版的，没有这个插件

L1cardo, 2022-05-10:
我也是刷过很多次固件，最新的总是有这个问题

lfdfxl, 2022-05-10:
插个楼，R2C也存在重启问题，连续开机1天22小时，今天早上9点自动重启了。

Goodman-vip, 2022-05-10:
R4S从4.27后的版本也是都有问题，第二天重启后网络都很慢，试了多次都不行，又刷回4.27来了

L1cardo, 2022-05-13:
今早发现pve下的x86也直接挂了，重启以后也不行，pve一直显示internal-error，然后控制台不能连接op

john5du, 2022-05-14:
我R2S更新了之后也是老死机重启，结果发现是cpu频率设置的高了。用cpufreq设置到1000上下就OK了，连续运行了几天了，没出问题。

L1cardo, 2022-05-20:
@john5du 你好 请问是这个插件吗？这里显示40万。。。
[image]

john5du, 2022-05-20:
> @john5du 你好 请问是这个插件吗？最小就是40万 [image] [image]

400khz 也太小了，我设置成1000khz上下

quietcoolwu, 2022-05-29:
> 我R2S更新了之后也是老死机重启，结果发现是cpu频率设置的高了。用cpufreq设置到1000上下就OK了，连续运行了几天了，没出问题。

@L1cardo 
@john5du 
請問這個問題解決了麼.... 調節頻率的方式能 work 嗎 還是必須回退

john5du, 2022-05-30:
> > 我R2S更新了之后也是老死机重启，结果发现是cpu频率设置的高了。用cpufreq设置到1000上下就OK了，连续运行了几天了，没出问题。
> 
> @L1cardo @john5du 請問這個問題解決了麼.... 調節頻率的方式能 work 嗎 還是必須回退

设置合理的频率能解决啊，我已经稳定运行一个月了。

yuanzhuohao, 2022-05-30:
> 

你设置的参数是怎样的？可以截图一下吗

yuanzhuohao, 2022-05-30:
> > > 我R2S更新了之后也是老死机重启，结果发现是cpu频率设置的高了。用cpufreq设置到1000上下就OK了，连续运行了几天了，没出问题。
> > 
> > 
> > @L1cardo @john5du 請問這個問題解決了麼.... 調節頻率的方式能 work 嗎 還是必須回退
> 
> 设置合理的频率能解决啊，我已经稳定运行一个月了。

你设置的参数是怎样的？可以截图一下吗

john5du, 2022-05-30:
> > > > 我R2S更新了之后也是老死机重启，结果发现是cpu频率设置的高了。用cpufreq设置到1000上下就OK了，连续运行了几天了，没出问题。
> 
> > > 
> 
> > > 
> 
> > > @L1cardo @john5du 請問這個問題解決了麼.... 調節頻率的方式能 work 嗎 還是必須回退
> 
> > 
> 
> > 设置合理的频率能解决啊，我已经稳定运行一个月了。
> 
> 
> 
> 你设置的参数是怎样的？可以截图一下吗
[image]

quietcoolwu, 2022-05-30:
> > 我R2S更新了之后也是老死机重启，结果发现是cpu频率设置的高了。用cpufreq设置到1000上下就OK了，连续运行了几天了，没出问题。
> 
> 
> 
> @L1cardo 
> 
> @john5du 
> 
> 請問這個問題解決了麼.... 調節頻率的方式能 work 嗎 還是必須回退

我發現我這邊的重啟問題是更新img md5不對導致的 用md5正確的版本+cpufreq後貌似解決了這個問題 跑幾天觀察下吧

L1cardo, 2022-05-31:
@yuanzhuohao 设置参数在这里https://github.com/klever1988/nanopi-openwrt/issues/1126#issuecomment-1132684118