stupidloud/nanopi-openwrt #1127: r2s with docker 0509版在线升级失败

lam2888, 2022-05-10:
[image]
己经换了闪迪A2性能的TF卡，不可能空间不够，也用读卡器修复过TF卡。sudo fsck.ext4 -y /dev/sdb1和sdb2 还是在线升级失败。

klever1988, 2022-05-10:
tmp分区占用过大，可能是某个插件占用的，升级时停掉不必要的插件再试

lam2888, 2022-05-10:
> 

试过了，只留下passwall，还是不行。

lam2888, 2022-05-10:
upg文件夹是干嘛的？tmp分区用的是不是tmpfs? 才850mb。
[image]

lam2888, 2022-05-10:
[image]
tmp分区小，应该是编译默认分区小吧？tmpfs应该3倍于解压后的固件大小才对。

lam2888, 2022-05-10:
[image]
r2s 只有1G运行内存，tmpfs用的是运行内存吧？默认开启tmpfs都用了350mb，在线升级r2s docker 0509版解压用了650mb，1G运存的tmpfs就不够用了。

lam2888, 2022-05-13:
[image]
用了隔壁家00575的在线升级固件，升级r2s with docker版，则没有问题，唯一的缺点是刷完后，需要ssh登录reboot.

closed 2022-05-16 by lam2888