stupidloud/nanopi-openwrt #1129: 请问R4S的CPU 性能优化调节设置怎么配置会比较合适？

choicky, 2022-05-12:
我最近买了R4S，刷了2022-05-11 自动发布的 [r4s-with-docker.img.gz](https://github.com/klever1988/nanopi-openwrt/releases/download/2022-05-11/r4s-with-docker.img.gz) 这个固件，目前正常，感谢大佬的奉献。

我最近发现 R4S 的温度比R2S低很多，看了一下，才发现 R4S的“CPU 性能优化调节设置”默认值比较保守，待机和Turbo Boost的CPU频率都是 `408000`，如图。

[image]

请问各位大佬，对于R4S，兼顾性能与稳定性的话，“CPU 性能优化调节设置” 的这几个参数设置为多少比较好呢？

- 待机 CPU 最小频率
- 最大 Turbo Boost CPU 频率
- CPU 切换频率触发阈值
- CPU 切换周期

moumou2015, 2022-05-13:
你运行个coremark就知道其实这个固件最高频率是2.2ghz/1.8ghz

choicky, 2022-05-14:
@moumou2015 5V2A的小米手机充电器作为电源，运行 `coremark` 后就死机了.... 
后来换了一个5V2.5A的才没死机。看来 R4S 不接外设的最大功耗超过10W呀。

closed 2022-05-17 by choicky