stupidloud/nanopi-openwrt #1137: 0515固件移动硬盘挂载失败

poorest1212, 2022-05-15:
从0423更新到0515，发现USB挂载失败，TTYD里手动mount sda2也不行

[image]

root@DJ-R2S:/mnt# mount /dev/sda2 /mnt/sda2
mount: mounting /dev/sda2 on /mnt/sda2 failed: Invalid argument

内核日志显示如下，不懂为啥会报错F2FS superblock找不到，这个移动硬盘挂载在mac上好好的

> Sun May 15 14:06:12 2022 kern.info kernel: [ 1894.931797] F2FS-fs (sda2): Magic Mismatch, valid(0xf2f52010) - read(0x41b05066)
> Sun May 15 14:06:12 2022 kern.err kernel: [ 1894.932452] F2FS-fs (sda2): Can't find valid F2FS filesystem in 1th superblock
> Sun May 15 14:06:12 2022 kern.info kernel: [ 1894.933646] F2FS-fs (sda2): Magic Mismatch, valid(0xf2f52010) - read(0x0)
> Sun May 15 14:06:12 2022 kern.err kernel: [ 1894.934241] F2FS-fs (sda2): Can't find valid F2FS filesystem in 2th superblock
> Sun May 15 14:07:35 2022 kern.info kernel: [ 1978.169280] F2FS-fs (sda2): Magic Mismatch, valid(0xf2f52010) - read(0x41b05066)
> Sun May 15 14:07:35 2022 kern.err kernel: [ 1978.169934] F2FS-fs (sda2): Can't find valid F2FS filesystem in 1th superblock
> Sun May 15 14:07:35 2022 kern.info kernel: [ 1978.171140] F2FS-fs (sda2): Magic Mismatch, valid(0xf2f52010) - read(0x0)
> Sun May 15 14:07:35 2022 kern.err kernel: [ 1978.171736] F2FS-fs (sda2): Can't find valid F2FS filesystem in 2th superblock
> Sun May 15 14:08:09 2022 daemon.warn odhcpd[924]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
> Sun May 15 14:08:38 2022 daemon.notice ttyd[2593]: WS closed from 192.168.2.169, clients: 0
> Sun May 15 14:08:38 2022 daemon.notice ttyd[2593]: killing process 21668 with signal: 1 (SIGHUP)
> Sun May 15 14:08:38 2022 daemon.notice ttyd[2593]: process killed with signal 1 (SIGHUP), pid: 21668
> Sun May 15 14:09:23 2022 daemon.err block: No "mount.exfat" utility available
> Sun May 15 14:09:23 2022 daemon.err block: mounting /dev/sda2 (exfat) as /mnt/sda2 failed (25) - Not a tty
> Sun May 15 14:09:45 2022 daemon.err block: No "mount.exfat" utility available
> Sun May 15 14:09:45 2022 daemon.err block: mounting /dev/sda2 (exfat) as /mnt/sda2 failed (25) - Not a tty
> Sun May 15 14:09:59 2022 daemon.err block: No "mount.exfat" utility available
> Sun May 15 14:09:59 2022 daemon.err block: mounting /dev/sda2 (exfat) as /mnt/sda2 failed (25) - Not a tty
> Sun May 15 14:12:42 2022 daemon.warn odhcpd[924]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
> Sun May 15 14:14:02 2022 daemon.err uhttpd[1054]: netstat: showing only processes with your user ID
> Sun May 15 14:14:03 2022 cron.err crond[23347]: crond (busybox 1.35.0) started, log level 8
> Sun May 15 14:14:04 2022 daemon.warn odhcpd[924]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
> Sun May 15 14:14:07 2022 daemon.err uhttpd[1054]: netstat: showing only processes with your user ID
> Sun May 15 14:14:09 2022 daemon.warn odhcpd[924]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
> Sun May 15 14:14:13 2022 cron.err crond[25080]: crond (busybox 1.35.0) started, log level 8
> Sun May 15 14:15:10 2022 daemon.err block: No "mount.exfat" utility available
> Sun May 15 14:15:10 2022 daemon.err block: mounting /dev/sda2 (exfat) as /mnt/sda2 failed (25) - Not a tty

klever1988, 2022-05-15:
再试试

poorest1212, 2022-05-16:
还是搞不定，exFAT模块会不会有问题？

> root@DJ-R2S:/mnt# lsblk -f
> NAME        FSTYPE FSVER LABEL  UUID                                 FSAVAIL FSUSE% MOUNTPOINTS
> sda                                                                                 
> ├─sda1      vfat         EFI    67E3-17ED                                           
> └─sda2      exfat        DJ-2T  D070-B665                                           
> mmcblk0                                                                             
> ├─mmcblk0p1 ext4         kernel 84173db5-fa99-e35a-95c6-28613cc79ea9                
> └─mmcblk0p2 ext4         rootfs ff313567-e9f1-5a5d-9895-3ba130b4a864   14.1G     3% /
> root@DJ-R2S:/mnt# mount -t exfat /dev/sda2 /mnt/sda2
> mount: mounting /dev/sda2 on /mnt/sda2 failed: No such device

lhf19900402, 2022-06-11:
我也是的  挂载不上了。。。。

poorest1212, 2022-06-18:
我最后是把硬盘给格式化成ext4了，估计是exfat的有问题