stupidloud/nanopi-openwrt #1138: 网络断连问题

wwdws1, 2022-05-15:
最近两周的固件都有问题，表现在ssh连接到外网的机器，几分钟就闪断一次，导致过几分钟就需要重新登陆ssh
目前在用的4.24fork后编译的没问题，系统日志如下
`Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_CONTAINER_OS_NAME=none
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_CONTAINER_OS_ID=none
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_CONTAINER_OS_ID_LIKE=none
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_CONTAINER_OS_VERSION=none
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_CONTAINER_OS_VERSION_ID=none
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_CONTAINER_OS_DETECTION=none
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_CONTAINER_IS_OFFICIAL_IMAGE=false
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_HOST_OS_NAME=OpenWrt
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_HOST_OS_ID=openwrt
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_HOST_OS_ID_LIKE=lede openwrt
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_HOST_OS_VERSION=SNAPSHOT
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_HOST_OS_VERSION_ID=snapshot
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_HOST_OS_DETECTION=/etc/os-release
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_HOST_IS_K8S_NODE=false
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_KERNEL_NAME=Linux
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_KERNEL_VERSION=5.4.192
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_ARCHITECTURE=aarch64
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_VIRTUALIZATION=none
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_VIRT_DETECTION=none
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_CONTAINER=unknown
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_CONTAINER_DETECTION=none
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_CPU_LOGICAL_CPU_COUNT=4
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_CPU_VENDOR=
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_CPU_MODEL=
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_CPU_FREQ=1608000
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_CPU_DETECTION=sysfs procfs
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_TOTAL_RAM=1038794752
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_RAM_DETECTION=procfs
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_TOTAL_DISK_SIZE=0
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: NETDATA_SYSTEM_DISK_DETECTION=sysfs
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: SQLite database /var/cache/netdata/netdata-meta.db initialization
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: SQLite database initialization completed
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: CONFIG: cannot load user config '/etc/netdata/stream.conf'. Will try stock config.
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: Host 'OpenWrt' (at registry as 'OpenWrt') with guid 'f8e2e31c-d420-11ec-8456-70b5e831bb7a' initialized, os 'linux', timezone 'UTC', tags '', program_name 'netdata', program_version '20200416-2054-g4e197f87a', update every 2, memory mode save, history entries 3996, streaming disabled (to '' with api key ''), health disabled, cache_dir '/var/cache/netdata', varlib_dir '/var/lib/netdata', health_log '/var/lib/netdata/health/health-log.db', alarms default handler '/usr/lib/netdata/plugins.d/alarm-notify.sh', alarms default recipient 'root'
Sun May 15 15:31:02 2022 daemon.err netdata[3615]: PROCFILE: Cannot open file '/etc/netdata/.environment'
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: Initializing command server.
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: thread created with task id 3870
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: set name of thread 3870 to BACKENDS
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: cleaning up...
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: thread with task id 3870 finished
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: thread created with task id 3871
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: set name of thread 3871 to EXPORTING
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: CONFIG: cannot load user exporting config '/etc/netdata/exporting.conf'. Will try the stock version.
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: thread created with task id 3864
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: set name of thread 3864 to PLUGIN[idlejitt
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: Initializing file /var/cache/netdata/system.idlejitter/main.db.
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: thread created with task id 3872
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: set name of thread 3872 to WEB_SERVER[stat
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: thread created with task id 3873
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: set name of thread 3873 to PLUGIN[timex]
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: thread created with task id 3869
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: set name of thread 3869 to STATSD
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: thread created with task id 3874
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: set name of thread 3874 to PLUGIN[tc]
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: thread created with task id 3875
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: set name of thread 3875 to PLUGIN[diskspac
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: thread created with task id 3876
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: set name of thread 3876 to PLUGIN[proc]
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: starting worker 2
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: starting worker 3
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: starting worker 4
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: thread created with task id 3868
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: set name of thread 3868 to SERVICE
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: No connector instances to activate
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: EXPORTING: no exporting connectors configured
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: cleaning up...
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: thread with task id 3871 finished
Sun May 15 15:31:02 2022 daemon.info netdata[3615]: thread created with task id 3865
[... 190 lines cut ...]
Sun May 15 15:35:00 2022 cron.err crond[5386]: USER root pid 7652 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sun May 15 15:40:00 2022 cron.err crond[5386]: USER root pid 8321 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sun May 15 15:45:00 2022 cron.err crond[5386]: USER root pid 9022 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sun May 15 15:49:48 2022 kern.err kernel: [ 1152.494245] r8152 4-1:1.0 eth1: skb_to_sgvec fail -90
Sun May 15 15:50:00 2022 cron.err crond[5386]: USER root pid 9702 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sun May 15 15:55:00 2022 cron.err crond[5386]: USER root pid 10392 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sun May 15 16:00:00 2022 cron.err crond[5386]: USER root pid 11056 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sun May 15 16:05:00 2022 cron.err crond[5386]: USER root pid 11744 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sun May 15 16:10:00 2022 cron.err crond[5386]: USER root pid 12413 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sun May 15 16:15:00 2022 cron.err crond[5386]: USER root pid 13092 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sun May 15 16:20:00 2022 cron.err crond[5386]: USER root pid 13769 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sun May 15 16:25:00 2022 cron.err crond[5386]: USER root pid 14452 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sun May 15 16:30:00 2022 cron.err crond[5386]: USER root pid 15127 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sun May 15 16:35:00 2022 cron.err crond[5386]: USER root pid 15826 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sun May 15 16:39:15 2022 daemon.info netdata[3615]: Initializing file /var/cache/netdata/ipv4.sockstat_tcp_mem/main.db.
Sun May 15 16:39:15 2022 daemon.info netdata[3615]: Initializing file /var/cache/netdata/ipv4.sockstat_tcp_mem/mem.db.
[... cut ...]
Sun May 15 16:40:21 2022 daemon.info netdata[3615]: Initializing file /var/cache/netdata/ipv4.tcphandshake/main.db.
Sun May 15 16:40:21 2022 daemon.info netdata[3615]: Initializing file /var/cache/netdata/ipv4.tcphandshake/EstabResets.db.
Sun May 15 16:40:21 2022 daemon.info netdata[3615]: Initializing file /var/cache/netdata/ipv4.tcphandshake/OutRsts.db.
Sun May 15 16:40:21 2022 daemon.info netdata[3615]: Initializing file /var/cache/netdata/ipv4.tcphandshake/AttemptFails.db.
Sun May 15 16:40:21 2022 daemon.info netdata[3615]: Initializing file /var/cache/netdata/ipv4.tcphandshake/TCPSynRetrans.db.
`

klever1988, 2022-05-15:
这跟路由应该是无关的吧，给sshd添加
ClientAliveInterval 和 ClientAliveCountMax 参数试试

wwdws1, 2022-05-18:
加了参数之后ssh没问题了，但是用vscode连接到服务器上的时候还有这个问题，而且确实存在刷到4月24日的固件就没问题，并且ToDesk也一直有断网提示，所以应该还是路由器的问题。配置都是用的4月24日刚备份配置还原的，是不是还原配置出的问题呢

faizulpaydo, 2022-06-04:
i also have same issue sir .. how to solve it?

Sat Jun  4 21:40:00 2022 cron.err crond[19687]: USER root pid 5759 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sat Jun  4 21:41:25 2022 authpriv.info dropbear[3420]: Exit (root) from <192.168.2.216:55325>: Error reading: Connection reset by peer
Sat Jun  4 21:41:31 2022 daemon.notice ttyd[7680]: HTTP / - 192.168.2.216
Sat Jun  4 21:41:31 2022 daemon.notice ttyd[7680]: HTTP /token - 192.168.2.216
Sat Jun  4 21:41:31 2022 daemon.notice ttyd[7680]: WS   /ws - 192.168.2.216, clients: 1
Sat Jun  4 21:41:31 2022 daemon.notice ttyd[7680]: started process, pid: 7664
Sat Jun  4 21:41:45 2022 auth.info login[7664]: root login on 'pts/0'
Sat Jun  4 21:45:00 2022 cron.err crond[19687]: USER root pid 8517 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sat Jun  4 21:49:50 2022 daemon.notice ttyd[7680]: WS closed from 192.168.2.216, clients: 0
Sat Jun  4 21:49:50 2022 daemon.notice ttyd[7680]: killing process 7664 with signal: 1 (SIGHUP)
Sat Jun  4 21:49:50 2022 daemon.notice ttyd[7680]: process killed with signal 1 (SIGHUP), pid: 7664
Sat Jun  4 21:50:00 2022 cron.err crond[19687]: USER root pid 9713 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sat Jun  4 21:55:00 2022 cron.err crond[19687]: USER root pid 12123 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sat Jun  4 22:00:00 2022 cron.err crond[19687]: USER root pid 14637 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sat Jun  4 22:05:01 2022 cron.err crond[19687]: USER root pid 15877 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sat Jun  4 22:10:00 2022 cron.err crond[19687]: USER root pid 17945 cmd /etc/init.d/oled restart >/dev/null 2>&1