stupidloud/nanopi-openwrt #1145: 大佬能不能麻烦添加一个 Socat 的 UI 包

gee1k, 2022-05-21:
因为用的移动宽带，没有 IPV4 公网，使用 IPV6 DDNS 访问，IPV6 的端口转发需要用到 socat，目前软件库里有 socat 的包，但是没有 UI。只能使用命令的方式。能不能麻烦大佬添加一下相关包

Soar1989, 2022-05-22:
试一下这个 opkg install luci-app-socat

gee1k, 2022-05-22:
> 试一下这个 opkg install luci-app-socat

仓库没这个包的

gee1k, 2022-05-22:
@klever1988 您好，我看到您添加了这个包，并发布了最新版 05-22。我在命令行在线升级过后，还是没有这个包呢

[image]

gee1k, 2022-05-22:
然后我想问一下，我卸载了一些不需要的插件。升级后这些插件还是会回来。请问能做到升级时不自动安装已卸载的插件吗？

klever1988, 2022-05-22:
忘记加参数了，现在加号重走~
另外那个重复安装插件的问题，你需要把语言包也卸载掉才不会再出现。
软件包已安装那里搜索luci-i18n

gee1k, 2022-05-22:
好的，谢谢。那我等下次更新

gee1k, 2022-05-22:
您好，反馈另一个问题：
DDNS 在系统重启后不会自动启动，每次重启需要手动开启

gee1k, 2022-05-25:
> 然后我想问一下，我卸载了一些不需要的插件。升级后这些插件还是会回来。请问能做到升级时不自动安装已卸载的插件吗？

@klever1988 你好，经过测试，发现固件内自带的插件即使相关的所有包都卸载了，更新版本的时候还是会安装回来

gee1k, 2022-05-25:
我上版本将这些插件全部卸载干净了，升级完都回来了。或者说这是不是就是正常的

[image]

gee1k, 2022-05-25:
> 忘记加参数了，现在加号重走~ 另外那个重复安装插件的问题，你需要把语言包也卸载掉才不会再出现。 软件包已安装那里搜索luci-i18n

然后上面这里说的这个 luci-app-socat 插件已经是找不到。
不知道是不是这个插件的问题，我现在是自己从这里下载的包手动安装的

https://op.supes.top/packages/aarch64_generic/luci-app-socat_git-22.138.56692-eb87df8_all.ipk

klever1988, 2022-05-25:
你用的不是slim啊？

socat我是加到slim的软件源里的，没有预装到其他版本

klever1988, 2022-05-25:
然后你想要不多余插件的话 升级的话需要升级slim

`wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | ver=-slim bash`

closed 2022-07-09 by gee1k