stupidloud/nanopi-openwrt #1147: 这几天折腾，经常遇到因为突然断电而ext4分区只读的情况，不知大佬以后是否会变异squashfs格式的？

choicky, 2022-05-23:
如题。

据说 `squashfs` 格式在意外断电后，也不会出现只读；而 `ext4` 格式有一定的几率，目测在 R4S 上游三四成的几率。

choicky, 2022-05-23:
刚才 fork 了一个，但不知道如何编译出 `squashfs` 格式的。求大佬指导

zhangguanzhang, 2022-06-05:
你有r2s吗，我做了r2s的squashfs格式固件，还支持在线升级扩容

choicky, 2022-06-07:
R2S 我也有。你的固件在哪呢？请分享一下固件，以及编译成 squashfs 的方法。

zhangguanzhang, 2022-06-07:
> R2S 我也有。你的固件在哪呢？请分享一下固件，以及编译成 squashfs 的方法。

https://github.com/zhangguanzhang/Actions-OpenWrt/releases/tag/test

choicky, 2022-06-11:
@zhangguanzhang 多谢大佬。

zhangguanzhang, 2022-06-11:
> @zhangguanzhang 多谢大佬。

详细信息写在这里了 https://github.com/zhangguanzhang/Actions-OpenWrt/tree/main#%E6%94%AF%E6%8C%81%E7%9A%84%E8%AE%BE%E5%A4%87%E5%88%97%E8%A1%A8

zhangguanzhang, 2022-06-12:
@choicky 用得还行吗，有啥建议

zephyrpathsofglory, 2022-11-03:
> 你有r2s吗，我做了r2s的squashfs格式固件，还支持在线升级扩容

有点不明白，squashfs 本身不就是只读的文件系统吗？现在就是要让系统变成 RW 的，为啥要刷个 只读的系统

zhangguanzhang, 2022-11-03:
> > 你有r2s吗，我做了r2s的squashfs格式固件，还支持在线升级扩容
> 
> @zhangguanzhang 有点不明白，squashfs 本身不就是只读的文件系统吗？现在就是要让系统变成 RW 的，为啥要刷个 只读的系统

是基础的rootfs是只读，然后利用overlay的diff，rootfs的读写都是overlay分区里，恢复出厂设置是把diff丢掉

zgc, 2022-11-12:
折腾了一个通宵，我也不知道怎么也触发了这个问题！！因为我频繁升级固件？