stupidloud/nanopi-openwrt #1148: 关于更新脚本

amwsuhje, 2022-05-24:
5-24没有更新到R4S，自动更新脚本不会找5-23更新的R4S更新，只会提示没更新

StephenLiz, 2022-05-29:
root@ImmortalWrt:~# wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | ver=-slim bash
Downloading file:///local_feed/Packages.gz
Updated list of available packages in /var/opkg-lists/local
Package coreutils-truncate (9.0-2) installed in root is up to date.
准备更新到2022-05-29
--2022-05-29 22:16:27--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2022-05-29/r4s-slim.img.gz
Resolving ghproxy.com... 146.56.158.111
Connecting to ghproxy.com|146.56.158.111|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 319762601 (305M) [application/octet-stream]
Saving to: 'STDOUT'

-                          100%[=====================================>] 304.95M   798KB/s    in 6m 38s

2022-05-29 22:23:06 (785 KB/s) - written to stdout [319762601/319762601]

--2022-05-29 22:23:06--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2022-05-29/r4s-slim.img.gz.md5
Resolving ghproxy.com... failed: Name does not resolve.
wget: unable to resolve host address 'ghproxy.com'

wao1201, 2022-05-29:
我遇到类似问题，感觉下载服务器不稳定
root@OpenWrt:~# wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | ver=-slim bash
Downloading file:///local_feed/Packages.gz
Updated list of available packages in /var/opkg-lists/local
Package coreutils-truncate (9.0-2) installed in root is up to date.
准备更新到2022-05-29
--2022-05-29 23:09:10--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2022-05-29/r2s-slim.img.gz
Resolving ghproxy.com... 144.24.74.73
Connecting to ghproxy.com|144.24.74.73|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 319636448 (305M) [application/octet-stream]
Saving to: 'STDOUT'

-                                  75%[=============================================>               ] 231.25M   445KB/s    in 61s

Cannot write to '-' (Invalid argument).
gzip: unexpected end of file
--2022-05-29 23:10:13--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2022-05-29/r2s-slim.img.gz.md5
Resolving ghproxy.com... 155.248.203.57
Connecting to ghproxy.com|155.248.203.57|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 50 [application/octet-stream]
Saving to: 'md5sum.txt'

md5sum.txt                        100%[============================================================>]      50  --.-KB/s    in 0s

2022-05-29 23:10:15 (3.72 MB/s) - 'md5sum.txt' saved [50/50]

2022-05-29固件已下载
固件HASH值匹配失败，脚本退出