stupidloud/nanopi-openwrt #1149: 添加LuCI的SSL组件

InspoOnU, 2022-05-25:
自行安装的时候提示内核版本不匹配，可以把luci-ssl添加进去吗

klever1988, 2022-05-25:
包名是啥

InspoOnU, 2022-05-25:
> 包名是啥

luci-ssl

klever1988, 2022-05-25:
没有这个包，你能不能提供一下错误内容

InspoOnU, 2022-05-25:
> 没有这个包，你能不能提供一下错误内容

opkg install luci-ssl
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libpthread and libpthread) providing same name marked HOLD or PREFER. Using latest.
Installing luci-ssl (git-20.244.36115-e10f954) to root...
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/luci/luci-ssl_git-20.244.36115-e10f954_all.ipk
Installing libustream-wolfssl20201210 (2022-01-16-868fd881-1) to root...
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/base/libustream-wolfssl20201210_2022-01-16-868fd881-1_aarch64_generic.ipk
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.115-1-2fda71a03d647cd5ecb63802c4826211) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.115-1-2fda71a03d647cd5ecb63802c4826211) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.115-1-2fda71a03d647cd5ecb63802c4826211) for kmod-nf-log
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-log found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.115-1-2fda71a03d647cd5ecb63802c4826211) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.115-1-2fda71a03d647cd5ecb63802c4826211) for kmod-nf-conntrack
 * check_data_file_clashes: Package libustream-wolfssl20201210 wants to install file /lib/libustream-ssl.so
	But that file is already provided by package  * libustream-openssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.

InspoOnU, 2022-05-25:
我把slim版下面注释的软件源也加上了，除了kmod以外的软件包都可以用。
是不是编译的时候选上luci-ssl就可以用了？

InspoOnU, 2022-05-26:
> 是的，但是fork项目没有编译，只是生成包的过程，你想重走编译的话需要去掉下面这个
> 

加入这几个软件包后就编译失败了

klever1988, 2022-05-26:
把CONFIG_PACKAGE_adguardhome=y拿掉应该可以

InspoOnU, 2022-05-26:
> 把CONFIG_PACKAGE_adguardhome=y拿掉应该可以

刷了镜像以后还是没有luci-ssl，那几个kmod也没有。opkg安装还是报了同样的错误。

klever1988, 2022-05-27:
CONFIG_PACKAGE_libustream-wolfssl=y
CONFIG_PACKAGE_CONFIG_PACKAGE_px5g-wolfssl=y

InspoOnU, 2022-05-27:
> 

加到common.seed里还是不行，似乎所有加到这个文件里的软件包都没效果。

klever1988, 2022-05-28:
https://github.com/coolsnowwolf/luci/tree/master/collections

L大的仓库没有这个包

closed 2022-05-28 by InspoOnU

klever1988, 2022-05-28:
你可以试一下luci-ssl-openssl看看能不能满足需要
主要还是要看你上层要装什么软件

InspoOnU, 2022-05-28:
> 你可以试一下luci-ssl-openssl看看能不能满足需要
> 主要还是要看你上层要装什么软件

这个可以，感谢