stupidloud/nanopi-openwrt #1151: r2s谷歌连接测试正常，但是不能翻墙

simpleyang1, 2022-05-29:
新刷的这个固件，然后订阅也正常更新了
是这个固件版本klever1988/nanopi-openwrt 2022-05-26
[image]
[image]
看起来都正常的，就是不能用，连接是猫----r2s----K3C，猫负责拨号，k3c做无线ap

simpleyang1, 2022-05-29:
[image]

cc9014, 2022-05-29:
先排除软件问题，试试passwall或者openclash看看，我昨天刚从骷髅头固件切换过来，也是0526的版本，用openclash一切正常。

poctopus, 2022-05-30:
0130版本之后的SSRP有问题，好像都没解决。