stupidloud/nanopi-openwrt #1154: 近半年多来的版本似乎对r2s越来越不友好了

lesliehua, 2022-06-01:
不知道有没有人跟我一样，在近半年多的时间里，更新已经不像以前那样稳妥，以前直接终端在线升级就好，基本上不会有什么大问题，但现在我已经不太敢升级了。
就好比今天升级了531的版本，升级完所有操作都不能保存，甚至连更改界面风格也不能成功保存。
然后再去刷530的，也是一样，并且修改设置后（比如更改passwall里的dns或模式选项）就直接error了，且无法再更改回去，只能重新刷。
再尝试了几次之后，我不得不选择今年0108的版本，那个版本是我当时用过感觉尚好时存下来的固件，以备不时之需。

所以我在想，是不是因为r2s的硬件相对比较老旧了，跟不太上大佬更新对硬件的需求？

klever1988, 2022-06-01:
大概率是文件系统损坏，终端内升级是直接绕过文件系统读写存储卡的，如果dmesg系统日志提示file system read-only就必须拔卡用工具重新刷固件才能修复，可以提前备份设置

lesliehua, 2022-06-02:
> 大概率是文件系统损坏，终端内升级是直接绕过文件系统读写存储卡的，如果dmesg系统日志提示file system read-only就必须拔卡用工具重新刷固件才能修复，可以提前备份设置

对，确实在第三次刷的时候看见了read-only的提示，但拔卡用balenaEtcher重新本地刷固件之后，还是不能修改设置们，包括0531和0530两天的版本

klever1988, 2022-06-02:
卡会不会寿终正寝了

lesliehua, 2022-06-04:
> 卡会不会寿终正寝了

这个就不好说了。现在暂时刷了1月8号备份的一个固件，目前为止OK
谢谢大佬～～