stupidloud/nanopi-openwrt #1158: 这个zerospace 没看懂是干啥的

zhangguanzhang, 2022-06-02:
```
cd lede/; pv /dev/zero > zerospace || true; sync; rm -f zerospace; cd -
```

klever1988, 2022-06-02:
把写过数据的块填0，这样btrfs就可以进一步把镜像压缩变小。

zhangguanzhang, 2022-06-02:
收到

zhangguanzhang, 2022-06-02:
[image]
[image]
另外咨询下，这个名字你是怎么搞的，我看了下你的yml和github action文档没看到那里能设置成不一样的

klever1988, 2022-06-02:
event_type

https://github.com/klever1988/nanopi-openwrt/blob/4a0f7a67d75682cbdd54b7f563499653b9fe4b7b/.github/workflows/dispatch.yml#L16

https://gist.github.com/Daniel-ltw/552b90800b07c22e4a83dfa68ada6318

zhangguanzhang, 2022-06-03:
感谢解答

closed 2022-06-03 by zhangguanzhang