stupidloud/nanopi-openwrt #1179: nano pi  r1  dd 到 emmc 开不了机

balainit, 2022-06-20:
dd 到 emmc 开不了机

[image]

官网固件  dd 到 emmc 能开机

[image]

请教这个有啥办法？谢谢☕️

balainit, 2022-06-21:
[image]

2020 年内的固件，都有带ROM，2021年，就没有带ROM了。所以 dd 进去，开不了机。
我尝试用 2020 年的，带ROM 的 dd 进去可以开机。

klever1988, 2022-06-21:
看这个
https://github.com/klever1988/nanopi-openwrt/tree/obsolete

balainit, 2022-06-21:
> 看这个
> 
> https://github.com/klever1988/nanopi-openwrt/tree/obsolete

看了一圈没看出来，新版本的固件好像都是 sd 卡，fat32 格式的 boot分区，所以 dd 进 emmmc 开不了机。新版本的固件编译要怎么配置类似传统的boot分区呢？这样 dd 进 emmmc 就可以开机了。

klever1988, 2022-06-21:
不能开机是啥现象，电源灯呢？

balainit, 2022-06-21:
> 不能开机是啥现象，电源灯呢？

电源灯红色的，不闪烁。一直红色。由于没有接显示器接口，看到界面。

balainit, 2022-06-21:
我试过把release下载的固件dd到 sd 卡。能正常开机使用。就是dd到emmmc 。电源灯一直红色，没反应。

xiayang0521, 2022-06-21:
玩电视盒子的经验  如果能在SD卡启动 可以在SD卡启动好 执行命令（似乎是openwrt-install之类的）安装到EMMC