stupidloud/nanopi-openwrt #1183: R1s H5 不断重启,内核打印如下

FirYuen, 2022-06-25:
内核打印如下
```
�!�[   16.669265] br-lan: port 1(eth1) entered disabled state
[   17.933158] br-lan: port 2(wlan0) entered blocking state
[   17.938478] br-lan: port 2(wlan0) entered disabled state
[   17.944133] device wlan0 entered promiscuous mode
[   17.953717] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   17.960277] br-lan: port 2(wlan0) entered blocking state
[   17.965586] br-lan: port 2(wlan0) entered forwarding state
[   17.971410] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   27.324808] fast-classifier: starting up
[   27.328869] fast-classifier: registered
[   27.332923] ------------[ cut here ]------------
[   27.337539] kernel BUG at target-aarch64_cortex-a53_musl/linux-sunxi_cortexa53/fast-classifier/fast-classifier.c:1892!
[   27.348224] Internal error: Oops - BUG: 0 [#1] SMP
[   27.353018] Modules linked in: fast_classifier(+) shortcut_fe_cm xt_FULLCONENAT pppoe ppp_async l2tp_ppp rtl8189es pppox ppp_mppe ppp_gen
[   27.353193]  lzo iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat asn1_decoder sch_tbf sch_ingres
[   27.440582]  gpio_button_hotplug btrfs xor zstd_decompress zstd_compress zlib_inflate zlib_deflate xor_neon raid6_pq lzo_decompress lzo_c
[   27.540365] CPU: 3 PID: 4515 Comm: insmod Not tainted 5.4.199 #0
[   27.546370] Hardware name: FriendlyARM NanoPi R1S H5 (DT)
[   27.551779] pstate: 40400005 (nZcv daif +PAN -UAO)
[   27.556589] pc : init_module+0x2e4/0x1000 [fast_classifier]
[   27.562182] lr : init_module+0x2d0/0x1000 [fast_classifier]
[   27.567751] sp : ffffffc010c6bb40
[   27.571063] x29: ffffffc010c6bb40 x28: 000000000000001d 
[   27.576390] x27: 000000000000001d x26: ffffffc0091291c0 
[   27.581704] x25: ffffffc009129078 x24: ffffffc009129000 
[   27.587011] x23: ffffffc009129148 x22: ffffffc009129400 
[   27.592335] x21: ffffffc010b217c0 x20: ffffffc009128158 
[   27.597647] x19: 0000000000000000 x18: 0000000000000000 
[   27.602955] x17: 0000000000000000 x16: 0000000000000000 
[   27.608269] x15: 5416010b13001b1f x14: ffffffc010b46de2 
[   27.613579] x13: 0000000000000000 x12: ffffffc010b46000 
[   27.618890] x11: ffffffc010aa7000 x10: 0000000000000010 
[   27.624197] x9 : 0000000000000000 x8 : 6574736967657220 
[   27.629504] x7 : 3a72656966697373 x6 : 0000000000000001 
[   27.634815] x5 : 0000000000000000 x4 : 0000000000000000 
[   27.640121] x3 : 0000000000000000 x2 : ffffffc010a9a000 
[   27.645436] x1 : ffffffc00f451000 x0 : ffffffc009105000 
[   27.650745] Call trace:
[   27.653203]  init_module+0x2e4/0x1000 [fast_classifier]
[   27.658427]  do_one_initcall+0x74/0x1c8
[   27.662292]  do_init_module+0x40/0x1e0
[   27.666063]  load_module+0x1cb0/0x21b0
[   27.669813]  __do_sys_init_module+0x1ec/0x2f8
[   27.674185]  __arm64_sys_init_module+0x18/0x20
[   27.678630]  el0_svc_handler+0x7c/0xd8
[   27.682376]  el0_svc+0x8/0x40
[   27.685351] Code: d0040820 aa0003e2 f9440000 b4000040 (d4210000) 
[   27.691453] ---[ end trace e5a162ad36a52f60 ]---
[   27.696069] Kernel panic - not syncing: Fatal exception
[   27.701289] SMP: stopping secondary CPUs
[   27.705210] Kernel Offset: disabled
[   27.708695] CPU features: 0x00002,00002000
[   27.712784] Memory Limit: none
[   27.715837] Rebooting in 3 seconds..

U-Boot SPL 2020.04 (Jun 23 2022 - 05:27:22 +0000)
DRAM: 512 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.4(release):20220505-91-g94e2af1bc-dirty
NOTICE:  BL31: Built : 05:27:22, Jun 23 2022
NOTICE:  BL31: Detected Allwinner H5 SoC (1718)
NOTICE:  BL31: Found U-Boot DTB at 0x408b5c0, model: FriendlyARM NanoPi R1S H5
NOTICE:  PMIC: Assuming H5 reference regulator design
NOTICE:  PSCI: System suspend is unavailable
```
貌似是fast-classifier这个内核组件有问题

FirYuen, 2022-06-25:
经过测试,fork一份后将common.seed中 `CONFIG_PACKAGE_kmod-fast-classifier=y`改为`CONFIG_PACKAGE_kmod-fast-classifier=n`后编译出的固件可以正常运行.
但是这个修改应该会改动所有项目,同时我暂时也不太清楚本项目的构建方式,所以可能无法merge一个可用的,只针对r1s的branch上来.

klever1988, 2022-06-25:
我只知道r1s的胖版固件会重启，slim正常，但slim也编译了fast-classifier。
可能是fast-classifier和某个内核模块在r1s上冲突了。

supperbo, 2022-06-26:
X86也出现重启！！！