stupidloud/nanopi-openwrt #1191: r2s线上更新404报错

polong, 2022-07-02:
wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | bash
Downloading https://mirrors.tencent.com/lede/snapshots/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_core
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_base
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_packages
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_luci
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_routing
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_telephony
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Package coreutils-truncate (9.0-2) installed in root is up to date.
准 备 更 新 到 2022-07-02
--2022-07-02 23:41:43--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2022-07-02/r2s.img.gz
Resolving ghproxy.com... 152.67.234.219
Connecting to ghproxy.com|152.67.234.219|:443... connected.
Unable to establish SSL connection.
gzip: invalid magic
awk: md5sum.txt: No such file or directory
root@ImmortalWrt:~# wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | bash
Downloading https://mirrors.tencent.com/lede/snapshots/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_core
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_base
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_packages
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_luci
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_routing
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_telephony
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Package coreutils-truncate (9.0-2) installed in root is up to date.
准 备 更 新 到 2022-07-02
--2022-07-02 23:44:47--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2022-07-02/r2s.img.gz
Resolving ghproxy.com... 152.67.234.219
Connecting to ghproxy.com|152.67.234.219|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2022-07-02 23:44:49 ERROR 404: NOT FOUND.
gzip: invalid magic
awk: md5sum.txt: No such file or directory

polong, 2022-07-02:
这样很多天了哦

amwsuhje, 2022-07-03:
7月2日没有r2s固件，他这个脚本只会找最新日期下的文件夹，找不到就404了，自己手动下载网页上传更新吧

polong, 2022-07-06:
噢噢明白了明白了，以前不是这么提醒的，没反应过来